AdaCore/Ada_Drivers_Library #3: Display drivers for '429-Discovery

Matthias-R, 2015-10-20:
Hi,

there are two different drivers for the display on the '429-Discovery board:

- in 'components' (ILI9341 - serial mode)
- in 'utils' (LTDC mode)

I think some hints in the file headers or in a readme file ('what do I find in this or that file') would be helpful.

BTW, the name of the directory 'utils' is somewhat misleading. I would expect some tools (e.g. a converter for character sets or something like that...) in a directory named 'utils'. The LCD related files are clearly drivers. I must admit that I haven't found a better directory name for the rest of the files yet.

pat-rogers, 2015-10-20:
Agreed, on both points.

The component driver is very low-level, and limited in scope. The LTDC interface does more, and will eventually be revised to be in terms of the component driver.

We considered "ancillary" instead of "utils" (among others) but have yet to find a good name.  It may be that we need to break up the contents to find a good name.

Thanks for the comments!

pat-rogers, 2015-10-21:
Probably I should move the STM32F4.* packages from utils/ to drivers/ and thereby simplify things a little. Originally the drivers/ dir was strictly for actual devices described by the RM, but that may be more narrow a definition than necessary.

Fabien-Chouteau, 2015-10-21:
LTDC

aethaniel, 2015-10-22:
hi, from what I see and understood about LTDC, the driver should be indeed in drivers.
But not in this current form as it seems to be a mix between low-level driver and middleware.
I would see the peripheral driver layer in 'drivers' and the layer using SPI and GPIO in 'utils' folder.

by the way, 'utils' isn't so badly named and is often used to place utility services which act in the inter-layers, ie not ll driver, not middleware but useful to app developer. Middleware are usually services like an usb stack, a filesystem management, a graphic layer etc...

pat-rogers, 2015-10-26:
Yes, the LTDC is a bit of a mixture at present.  It was written prior to the ILI9341 component driver so we envision it being revised accordingly, at some point, and made more independent of the board.

I agree that utils is a good name as long as that's what is in there.

lambourg, 2016-04-26:
The LTDC driver (and more generally framebuffer display) have been greatly refactored in the 'svd' branch. In particular:

- stm32-ltdc.ads now only configures the LTDC peripheral
- stm32-dma2d also only configures the peripheral
- a new HAL.Framebuffer interface has been introduce to handle framebuffer displays
- this interface is partially implemented in boards/common_stm32/ltdc/framebuffer_ltdc.ads.
- a new HAL.Bitmap type has also been introduced to perform operations on bitmap surfaces,
- a specialized version bitmap_dma2d allows hardware acceleration.
- all those interfaces are fully implemented at the board levels.

lambourg, 2016-04-27:
This is now integrated into the master branch. So I'm closing this issue, feel free to open a new one if you see ways to improve the current architecture  (even better if you can contribute changes: we're very open to your participation to this project)!

Best regards,
- Jerome

closed 2016-04-27 by lambourg