AdaCore/Ada_Drivers_Library #4: Seperation of register definitions and convenience functions

Matthias-R, 2015-10-20:
Hi,

the files in the 'drivers' directory contain both register definitions and convenience functions/procedures to access them. That's fine if these files are completely written by hand.
Considering automatic (at least partially automatic) generation of register definitions from CMSIS-SVD files, it could be beneficial to separate the definitions and the convenience functions / low level drivers.

pat-rogers, 2015-10-20:
That approach (automatic generation) seems inapplicable in this case, though, does it not?

pat-rogers, 2015-10-21:
Well, in the immediate case I think we are in hand-generated mode, but what are the possibilities down the road that you see?

pat-rogers, 2015-10-21:
What would you think of having the package private part generated, containing the register declarations?  Or if not the actual package text, maybe just generate the text of the register decls themselves, and then we could manually copy/paste them into a package declaration.

Matthias-R, 2015-10-21:
All registers of a specific processor are described in its CMSIS-SVD file. Unfortunately, some fields of the description are optional. If all optional information were there, a fully automatic generation of .ads-files with the register definitions would probably be possible. But even with not-so-complete SVD files (e.g. definitions of enumerations are completely missing in SVD files from STM) a fair amount of tedious work can be done automatically. My proof-of-concept SVD to ads converter generates a skeleton with record definitions for the registers. It leaves some manual work with type definitions for the fields in the register (especially if the type should be an enumeration - as I mentioned before, enumerations are missing in the files from STM). Some special constructs in the SVD files aren't recognized at all - but as I said, it is a proof of concept, far away from being complete.
I already had some discussions about the topic 'SVD to Ada converter' with Quentin. I'm not sure if this is the right place to discuss details of such a converter or if there should be another place?!

Regarding your last comment: Copy/paste could be a possibility if there is manual work to do anyway.
Nevertheless, I would prefer a clean separation between the register definitions (= the part with a direct correspondence to the UM of the processor, ideally autogenerated, in reality perhaps with some manual finishing) and functions/procedures based on these definitions. Even if this means that someone has to find even more sensible package names... ;-)
I don't like the idea of hiding the definitions in a private part. I would keep the possibility to access the registers directly - using the convenience functions should not be mandatory.

pat-rogers, 2015-10-22:
Yes, you should work with Fabien. 

I would very much welcome automatic generation, even if incomplete. Deciding whether or not to generate it within the text of a package declaration is a secondary matter.

lambourg, 2016-04-24:
Hello Matthias,

The transition of this drivers repository to make use of the SVD branch is now almost complete. Only one step left (merge the 'svd' branch with the 'master' branch). Il will thus close this Issue.

Best regards,
- Jerome

closed 2016-04-24 by lambourg