AdaCore/Ada_Drivers_Library #9: zero footprint library

huegene, 2016-02-09:
I don't really know where else to put this or where to look for answers but, I have experimented with porting the stm32f4 runtime library (sfp) to an stm32f103 device. By changing various stack sizes I was able to fit it all in the 20KB of RAM. Considering even smaller controllers it seems reasonable to go with the ZFP runtime systems. But i was not able to find any resources on how to implement interrupts with a ZFP rts. Can anyone help me out tackling this or even better provide me sources for study?
Note i've created my ZFP rts from the bb-runtime sources from libreada

Fabien-Chouteau, 2016-02-11:
On Wed, Feb 10, 2016 at 12:15 AM, huegene <notifications@github.com> wrote:

> I don't really know where else to put this or where to look for answers
> but, I have experimented with porting the stm32f4 runtime library (sfp) to
> an stm32f103 device. By changing various stack [...]
By definition with the ZFP you are on your own, so an interrupt is just a
call to a function (more or less). You can have a look at how we do it in
the ravenscar run-time in the handler.S file.

-- 
Fabien Chouteau

simonjwright, 2016-02-26:
See also Maciej Sobczak’s [Ada and SPARK on ARM Cortex-M](http://www.inspirel.com/articles/Ada_On_Cortex.html) - chapter 13.

Fabien-Chouteau, 2016-02-26:
Thanks @simonjwright, that a good pointer indeed.

@huegene I think you have enough info I will close this issue.

closed 2016-02-26 by Fabien-Chouteau