AdaCore/Ada_Drivers_Library #11: Implementing UART Streams

nesos, 2016-02-12:
Hi,

is it reasonable to have a stream interface for the UARTs ? That would ease the use of the UARTs and serialization. On the other hand streams are only available with the full Ravenscar RTS and it looks like SFP doesn't support streams.
Are there any thoughts about this?

pat-rogers, 2016-02-12:
Streams would be a useful interface.

There's no requirement that all facilities work with all the runtimes, although so far, at this lowest level, we have ensured they do. At higher levels more advanced features are reasonable, such as tagged types.

pat-rogers, 2016-05-21:
There is now a streams-oriented serial port abstraction in the examples/serial_ports project. (There are two other serial port abstractions as well.)

closed 2016-05-26 by pat-rogers