AdaCore/Ada_Drivers_Library #12: Ethernet & TCP/IP

oli-obk, 2016-03-10:
Are there any plans on making use of the Ethernet port on the stm32f7 discovery board?

Fabien-Chouteau, 2016-03-10:
It's definitely on the TODO list, please tell us if you start working on it.

lambourg, 2016-04-26:
If you look at the svd branch, there's examples/ethdemo that starts implementing some support for it. You can start from that and add what's missing.

lambourg, 2016-04-27:
The SVD branch has now been merged in master. So indeed, if you want to contribute, please have a look at components/eth.

This has not been integrated yet with the general framework as it's still a very early stage driver, but that's some basis you can start with.

Best regards,
- Jerome

closed 2016-04-27 by lambourg