AdaCore/Ada_Drivers_Library #17: SVD error for F7

astro-jerry, 2016-04-19:
In the svd directory, the file stm32_svd-spi.ads does not have the correct bit layouts for some of the registers (CR1, CR2, SR) - they seem to be for the F4 chips.  On the F7 these registers have changed.
Reference: page 1110 of RM0385 for STM32F75xxx and STM32F74xxx advanced ARM-based
32-bit MCUs

Fabien-Chouteau, 2016-04-19:
Hello,

We already have a "fixed" version of the STM32F7 SVD file in the svd2ada project:
https://github.com/AdaCore/svd2ada/tree/master/CMSIS-SVD

Please submit a fix there, generate the files and then submit a patch with fixed generated file on this repo.

Regards,

lambourg, 2016-04-24:
Hello, some additional definitions for the F7 SPI interface have been added. If you're spotting other issues like this one, please fill free to modify the SVD file from svd2ada and submit a fix. You can try the generation of the bindings using the makefile located in ARM/STM32/svd.

closed 2016-04-24 by lambourg