AdaCore/Ada_Drivers_Library #18: Driver depends on Board?

simonjwright, 2016-04-20:
drivers/stm32-sdram.adb depends on boards/stm32f429_discovery.ads: is this right?

lambourg, 2016-04-20:
As previously stated by Fabien, this branch is now considered obsolete. Most of the work is now performed on the 'svd' branch where STM32.SDRAM is not considered anymore as part of the drivers section, but as part of the board/components

closed 2016-04-20 by lambourg

simonjwright, 2016-04-20:
>As previously stated by Fabien, this branch is now considered obsolete.

Oh. Missed that.

Odd for the _master_ branch to be obsolete.

lambourg, 2016-04-20:
It's going to be fixed soon ;)

Matthias-R, 2016-04-20:
Regarding your first question: Yes, I think that's right. It is not really a generic driver for any type of SDRAM connected to the FMC, it's tailored to the SDRAM fitted on the discovery board.