AdaCore/Ada_Drivers_Library #25: Short Feedback

elgor, 2016-06-06:
Hi all, I want to share my first experiences with the library.
For my project I only use the Peripheral drivers and thus only included ./ARM and ./HAL.
So far I have used the following modules: GPIO, I2C, UART, SPI.

First of all, thank you for the great work the library works most of the time as expected!

However, I noticed some issues and/or have some suggestions for improvement.
Maybe some of these issues are already on your todo list or you have a special reason for it, but since there is no public list I will mention everything that came to my mind.

### GPIO
1. **No in out paramters if it is not necessary**
	In `procedure Set (Point : in out GPIO_Point)` from HAL-gpio.ads. If Point is in out then one has to declare a variable before calling the procedure and cannot use a constant value or return value.

2. **Are limitted types necessary for Peripheral descriptors?**
	I would like to write a mapping function that maps from my own device ID type e.g. (Baro, IMU) to the corresponding SPI Peripheral. However when I try to return a constant Peripheral object such as `STM32.Device.SPI_1` the compiler correctly complains that I cannot copy a limited type. It would be nice to have a type that identifies the peripheral without representing it by the actual address as the limited type does.

3. **Ambigious enum values**
I noticed that when I want to access a GPIO Port A for the `Set` function and type `STM32.Device.GPIO`, GPS suggests auto-completion for `GPIO_PORT_A` and `GPIO_A`  and it is not easy to see the difference between them.

### General Naming Conventions
The library will only be written once but used by many people who will read the source code and have to understand it. Thus a primary goal for naming should be readability and consistency. Even names become longer, they will be auto-completed by the IDE. I think the naming is quite good but I have some additional suggestions.

1. **Only common abbreviations:**
When I first saw `CRC_Poly` I though of CRC_Policy. Of course polynom makes much more sense but I would avoid abbreviations if they are not very very common or could be ambiguous. Depending on one's background different associations could come to mind.

2. **Types could have a postfix `_Type`**
I have seen this often in examples and use it myself. Advantages would be
 - fast identification of type identifiers
 - singleton naming for parameters: `Configure : Configure_Type`, no need to come up with a different name/abbreviation
 - developers might be already used to it

3. **Consistent parameter names**
I noticed that the I2C and SPI modules have different parameter names for the configure method. `I2C.configure` requires a `Handle` while `SPI.configure` requires a `Port`. It would be nice if the parameter names of equivalent subprograms are consistent.

4. **Consistent prefix**
Within SPI module almost all types start with the prefix `SPI_`, except for `Internal_SPI_Port`. Maybe it should be renamed to `SPI_Internal_Port` for consistency.

5. **No mixing of similar words such as `send` and `transmit`**
SPI has send and transmit as public subprograms and it is not easy to see which one to use. I2C has only transmit functions, so maybe it would be possible to remove the plain send procedure from the public part in the SPI module.

6. **Documentation of naming conventions**
Whatever the naming rules are, they should be documented somewhere (e.g. in the git wiki), so that new users can understand the reason behind the names and getting used to the library. This is beneficial

  1. for contributors to be able to follow the rules
  2. for users to understand code better/faster

### USART

1. **Unified Bus procedures**	
 I really like the Configure(Port, Configuration) methods.
Right now USART does not use a single configuration procedure like SPI/I2C do but has several. I think it would be more consistent if every Bus is configured with one configure procedure.

 Every Bus (SPI, I2C, UART) should have a transmit, receive and transfer/transceive method.The USART transmit procedure does not check the TXE flag and the receive method does not check RXNE or timeout as SPI/I2C do. Again I think it would be consistent if every transmit/receive procedure behaves the same.

2. **Clarify the unit of parameters**
	In `Set_Baud_Rate` the parameter `Baud_Rates` is of type word. It is not clear what the unit of this type is. Maybe `Baud_Rates` should be renamed to `Baud_Rate_In_Hz`

3. **Required order of call**
	I noticed that `Set_Baud_Rate` reads `CR1.OVER8` so a user should call `Set_Oversampling_Mode` beforehand.
	It is not clear from the interface that `Set_Baud_Rate` reads this register.
	This issue could also be avoided by a single configuration call.

### SPARK
I also want to use SPARK but the library is not in SPARK which I guess is not avoidable. However is it possible to shift the non SPARK parts in the body to be able to declare the spec with SPARK mode? Are there any plans or recommendation how to use the library in a SPARK project?

I hope this feedback is somehow useful for you. As I said it is just everything that came to my mind. I really like the library so far and it is very useful. So thank you for all the effort!

Kind regards,
Emanuel

Fabien-Chouteau, 2016-06-09:
Thanks for your feedback Emanuel! I think we need a little bit of time to process all that ;)

lambourg, 2016-08-04:
Hello Emanuel,

First of all, many thanks for your detailed feedbacks. It's both very helpful and very constructive :) Here is some elements answering partially your remarks, on the GPIO front:

1. I removed the need for in out mode recently (yesterday), as indeed this does not really make sense in real life, even though from a theoretical point of view and from a software point of view it would seem logical to have the GPIO point instance modified after a call to set, clear or toggle.

But practically and from someone used to drivers programming, this is indeed unexpected and limitating.

2.  On the limited type side for peripheral, note that those limited types are all aliased, which means that you can pass on access values when you need to reference them.

Having limited types is useful sometimes when we need to add values to the peripheral descriptor (such as peripheral states), whose values are not directly derived from the hardware registers. We need to ensure that there's only one instance describing the peripheral in such case. So for homogeneity purpose (as well as allowing this in future updates), we decided to have all peripherals described as limited aliased types.

3. Now on the name clashes, I removed the GPIO enum type, as it was a bit useless. The only useful situation is to retrieve the GPIO port number to setup external interrupts.

New answers to come later on your other points ;)
Best regards,
- Jerome

Fabien-Chouteau, 2016-08-04:
About GPIO 1. (in out paramters)

I think I already explained why I'm against removing the "in out" descriptors, but here are my arguments:

- We should use the language features to make the best possible interface. Using the 'out' descriptor shows that setting or clearing and IO may modify it. It's an information that we provide to the user of that interface.

- You may want to give a read-only 'access' to some part of the code. If we remove the 'out' descriptor we lose that control. Setting or clearing an IO means changing its state, which requires the right to modify it.

- For IO expenders like the MCP23008, one may want to keep an internal representation of the IO to avoid some costly communication with the device. This internal state requires modification rights. Example:

```
PA0.Clear; --  Clear the IO
PA0.Set; -- Set the IO
PA0.Set; -- Redundant,  this operation can be optimized, i.e. do nothing because the IO is already set
```

For your return value case @elgor,  I would return a "not null GPIO_Point_Ref".

lambourg, 2016-08-04:
My feeling is that such caching issue should be performed then at the GPIO_Port level, not at the GPIO_Point. In GPIO_Point, you have an access to the GPIO Port, so can do whatever you want there, ensuring coherent read/write (as the GPIO_Port is a limited type), as opposed to the GPIO_Point that should really be non-limited here.

I would have:

HAL.GPIO: 

type GPIO_Point is interface;
prodecure Set (Point : GPIO_Point) is abstract;
...

STM32.GPIO:

type GPIO_Port is limited private;

type GPIO_Point is new HAL.GPIO with record
   Port : access GPIO_Port;
   Pin  : GPIO_Pin;
end record;

Fabien-Chouteau, 2016-08-04:
Indeed that's one way to implement it, it doesn't mean we should restraint other ways and it's still hiding the fact that there's a modification of internal state.

On the other hand, I don't think there's a good reason to remove the 'out' descriptor.

lambourg, 2016-08-04:
Well, to me you're manipulating an identifier when you manipulate a GPIO_Point, not an actual representation of the hardware pin. In such case, one may expect all manipulations to be 'in', and the type being non-limited (such as me but your mileage may vary here of course). Although the GPIO point is a record type, I really see it as an enum value: just a label.

On the more practical side: having GPIO_Point as non-limited read-only values would allow their storage in structures you don't always have write access to, more flexibility in their use in general.

Fabien-Chouteau, 2016-08-04:
Considering GPIO_Point as identifier you are missing point 1 and 2 of my arguments and you are not using the nice features of the language (which is an important goal of this project).

Note that I'm mostly interested in HAL.GPIO.GPIO_Point here. GPIO_Point is an interface type so you can't put it directly a record anyway.

lambourg, 2016-08-07:
Well, having enum identifiers is one of the nice features of the language ;)

Next, it turns out that when you have the actual GPIO points (e.g. you have an application that knows perfectly well on what hardware it runs on, so have GPIO point as a constrained type), you really need those points to be identifiers: so they should not interfere with your own access rights you want to define for your device. 

You don't want read operations to be performed in "in out" mode just because we decided that pins should be "in out" whatever the use case, imho, but the read operation needs sets/clears to be performed on pins. In particular when we all know that it's not needed at all from an implementation point of view.

Fabien-Chouteau, 2016-08-08:
Sure, enums have their own benefits, but what you describe is a project specific and not portable handling of IO. You define an enum with meaningful identifiers for the IO in the context of a specific project. It doesn't apply in this library because we want to favor code reuse and portability with the HAL. Enums don't work in HAL.

closed 2018-06-01 by Fabien-Chouteau