AdaCore/Ada_Drivers_Library #28: EthDemo constant restart

jorgie0, 2016-07-11:
Hi,
I've got blink.adb running on a Nucleo STM32F429ZI board.

When I attempt to run Ethdemo.adb the program constantly resets.

It resets on line 160 of stm32-eth.adb
`Rx_Descs (I) :=
        (Rdes0 => (Own => 1, others => <>),
         Rdes1 => (Dic => 0, Rbs2 => 0,
                   Rer => (if Last then 1 else 0),
                   Rch => 1, Rbs => Rx_Buffer'Length,
                   others => <>),
         Rdes2 => W (Rx_Buffs (I)'Address),
         Rdes3 => W (Rx_Descs (I + 1)'Address));`

within the `procedure Init_Rx_Desc (I : Rx_Desc_Range)`

The Nucleo board uses a PHY LAN8742A-CZ-TR

Fabien-Chouteau, 2016-07-11:
Congrats for the blinky! That's a good first step :)

About the Ethernet support, what we have in the repository is only experimental and was only tested for the STM32F7 disco. So you will probably have to implement a few things for the 429 nucleo. 

The reset that you see probably comes from an unhandled exception, try to put a breakpoint on all exceptions with "(gdb) break exceptions" or stop the CPU before it resets "(gdb) break __gnat_last_chance_handler".

Note that you can define your own last chance handler, for example to blink an LED instead of reseting the board. You will find example of that in the repository.

jorgie0, 2016-07-14:
Hi
Thanks for the reply.
I found that the offending clause was in this line of code:
`    Rx_Descs (I) :=
        (Rdes0 => (Own => 1, others => <>),
         Rdes1 => (Dic => 0, Rbs2 => 0,
                   Rer => (if Last then 1 else 0),
                   Rch => 1, Rbs => Rx_Buffer'Length,
                   others => <>),
         Rdes2 => W (Rx_Buffs (I)'Address),
         Rdes3 => W (Rx_Descs (I + 1)'Address));`

With the actual issue being:
`Rx_Descs (I + 1)'Address`
For some reason this was being allocated an address of 0. 

I did put in an exception handler but that did not get executed, instead it dropped out to last chance handler.
`   exception
      when Constraint_Error =>
         GNAT.IO.Put ("Constraint");
      when others =>
         GNAT.IO.Put ("Something Else");
   end Init_Mac;`

Now the funny thing is that even after putting a dummy variable in to move Rx_Descs up I do not get any received packets.

I'm new to Ada on STM32s so I have not been able to find out which IP address gets allocated so unable to send any packets to the board.

Any suggestions.

jorgie0, 2016-07-16:
I've "solved" the exceptions not being propagated. Ravenscar does not propagate exceptions.
Damn.

I've also worked out that there is no easy answer to the TCP/IP stack.

So at this stage the program will run, if I put in the dummy variable to move the address of Rx_Descs up, but I'm not sure if it is correct. 

Why would there be a problem having Rx_Desc at address of 0?

lambourg, 2016-07-16:
Hello,

As Fabien already mentioned, the fact that the stm32.Eth package is placed in the examples directory and not in the usual drivers place means that this package is very experimental. It is meant to run only on the STM32F7-disco board for the moment, so some hard work will be needed to make it work on another board. Also this driver is meant to control the hardware of the board, not to provide a full IP stack. The Ada_Drivers_Library do not contain such IP stack (for the good reason that we don't have a proper ethernet driver yet).

Now on your specific questions:
* the ravenscar-sfp profile is a small foot-print runtime, that is equivalent to the zfp with just the ravenscar tasking added (zero-footprint: no runtime is embeddded in your code unless you explicitely call runtime functions). Zero footprint or small footprint implies no exception propagation is supported (this implies embedding code from the runtime. If you need exceptions, you can use the ravenscar-full profile, that contains much more functionalities (but embeds more code, so your application will be bigger).
* the null value for Rx_Desc: if you look at the code, Rx_Desc needs initialization and SDRAM. It is initialized at line 225, via a call to STM32.SDRAM.Reserve. Having null pointers is in general a problem and trying to access such address will generate a hardware fault (thus will reset the board).

closed 2016-07-16 by lambourg