AdaCore/Ada_Drivers_Library #30: Example: b__balls_demo.adb linking error, liblto_plugin.so missing

fustbariclation, 2016-08-05:
I'm missing a configuration step somewhere.

The demo task b__balls_demo.adb complies, but, at link time gives this error:

arm-eabi-gnatbind balls_demo.ali
arm-eabi-gcc -c b__balls_demo.adb
arm-eabi-gcc fractals_demo.o -Wl,--defsym=__stack_size=32768 -Wl,--gc-sections -Wl,--print-memory-usage -o fractals_demo
arm-eabi-gcc: fatal error: -fuse-linker-plugin, but liblto_plugin.so not found

A fix for this is documented here:

https://www.lpcware.com/content/forum/libltopluginso-not-found

However, this doesn't work. It looks as if something is missing from config.gpr, pointing to the library.

I'm using the example file stm32f7disco_examples.gpr  so my build line is:

gprbuild stm32f7disco_examples.gpr 

I'm using:

Linux host 4.4.0-31-generic #50-Ubuntu SMP Wed Jul 13 00:06:14 UTC 2016 i686 i686 i686 GNU/Linux

GPRBUILD GPL 2016 (20160515) (i686-pc-linux-gnu)

arm-eabi-gcc (GCC) 4.9.4 20160426 (for GNAT GPL 2016 20160515)

fustbariclation, 2016-08-07:
I was wondering if anybody had any thoughts about what the problem is. I'd grateful for any suggestion of things to try.

I do find this puzzling, as I'm sure that the demos have all been tested before.

lambourg, 2016-08-07:
Note that ubuntu is not part of the supported linux targets.

Now, what is the output of: ldd <GNAT install>/libexec/gcc/arm-eabi/4.9.4/liblto_plugin.so ?

fustbariclation, 2016-08-08:
Thank you for letting me know - I thought that linux was supported, I'd not known Ubuntu wasn't.

gprbuild doesn't work on the Raspberry Pi, and it doesn't work on the Mac, so Ubuntu seemed the next best bet.

There isn't a libexec directory, but:

ldd ./Ada_Drivers_Library/examples/liblto_plugin.so
	linux-gate.so.1 =>  (0xb7740000)
	libc.so.6 => /lib/i386-linux-gnu/libc.so.6 (0xb755b000)
	/lib/ld-linux.so.2 (0x800a7000)

ldd /media/sf_Ada/Ada_Drivers_Library/examples/balls/liblto_plugin.so
	linux-gate.so.1 =>  (0xb772d000)
	libc.so.6 => /lib/i386-linux-gnu/libc.so.6 (0xb7548000)
	/lib/ld-linux.so.2 (0x80061000)

This looks wrong to me - this should be an arm-eabi library, shouldn't it?

So there's a problem with the build. Is that correct?

Actually, looking at the install readme, I see ( in arm-elf-linux/adagpl-2016/gnatgpl )
"
GNAT GPL 2016 - GNAT compilation system

This version of GNAT GPL is officially supported on the
following hosts. Please see the "Special Notes" section
immediately following the targets section for special
instructions relevant to specific platforms.

PC/x86 and x86_64
  - Windows Vista Business, 7, 8, 8.1, and 10
  - Windows Server 2003, 2008, 2008R2 and 2012
  - Red Hat Enterprise Linux 5, 6 and 7
  - SuSE Linux Enterprise Server 11 and 12
  - Ubuntu 14.04 LTS, 16.04 LTS
  - Mac OS X Mavericks, Yosemite, El Capitan
"

lambourg, 2016-08-08:
Ah, fine, that's new indeed.

Concerning your issue, it has nothing to do with the Ada Drivers library, but has to do with your gnat installation: liblto_plugin.so is a gcc plugin, so is a host library.

I really don't know where your liblto_plugin file present in your example directory comes from, but after a proper gnat installation it should be present in (gnat install dir)/libexec/gcc/arm-eabi/4.9.4/liblto_plugin.so

Fabien-Chouteau, 2016-08-08:
@fustbariclation I can confirm that GNAT GPL 2016 is working on Ubuntu (version below), so this must be a installation/setup issue. 

liblto_plugin.so is an x86 library used by the compiler, you should not try to put it in your ARM project files. There must be a reason why the compiler can't find it.

I will close this issue because it's outside the scope of this project.

`chouteau@: ~/
-> lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 12.04.5 LTS
Release:        12.04
Codename:       precise
chouteau@: ~/
-> uname -a 
Linux [...] 3.5.0-43-generic #66~precise1-Ubuntu SMP Thu Oct 24 14:52:23 UTC 2013 x86_64 x86_64 x86_64 GNU/Linux
chouteau@ ~/
`

closed 2016-08-08 by Fabien-Chouteau