AdaCore/Ada_Drivers_Library #31: STM32F42x has UART 7, 8

simonjwright, 2016-08-07:
ARM/STM32/devices/stm32f42x/stm32-device.ads should support UART7, UART8 (RM0090 Fig 27 shows they are on STM32F42xxx and STM32F43xxx).

Also the AFs ...

lambourg, 2016-08-07:
Hi Simon,

As we're currently working on other parts of the drivers library, feel free to contribute patches to enhance the library. We're very open to contributions here... And the more contributions, the better. Moreover, adding drivers we're not actually testing is of very low interest, so if you have a situation where UART7/8 are needed, it's better if you add the proper driver to those peripherals.

Best regards,
- Jerome

lambourg, 2016-08-09:
Ok, this is now fixed!

As always, thanks for your report, Simon.

closed 2016-08-09 by lambourg

simonjwright, 2016-08-09:
Thanks, Jerome!