AdaCore/Ada_Drivers_Library #41: Ethernet - DHCP - MAC address - stm32f7disco

fustbariclation, 2016-08-23:
Has anybody, ever, enabled the Ethernet card for DHCP, and set its MAC address, using Ada?

Is there, possibly, somewhere, something like the echo example shown here:

http://www.st.com/resource/en/user_manual/dm00103685.pdf

If so, would it be possible to look at the code?

I can see how to do this in C, it’s easy, and fairly well documented here:

http://stm32f4-discovery.net/2015/02/library-52-ethernet-peripheral-on-stm32f4xx/

I have managed to find documentation for the register acronyms here:

https://github.com/deiKruve/openPowerLink-Ada/blob/master/stm32f-definitions/workfiles/stm32f4-o7xx-eth.ads.nopp

Unfortunately, this makes it clear that the bit (and it is only one bit, in one register) that sets the card to DHCP, and the place to put the MAC address in the initialisation code, are not there.

Any suggestions very gratefully received!

Fabien-Chouteau, 2016-08-24:
Tristan started to play with the Ethernet controller already. You can have a look here: https://github.com/AdaCore/Ada_Drivers_Library/blob/master/components/eth/stm32-eth.adb

fustbariclation, 2016-08-26:
> master/components/eth/stm32-eth.adb

Yes, thank you for that - I'd seen the program. I managed to find another
that had the abbreviations for the registers explained.

Unfortunately this does not include the two critical things to get the
ethernet card working:

- setting the MAC address
- Enabling DHCP

Once these are working, the rest should work well too - mostly, there's no
need to set the register details because they are negotiated by both sides.

Do you know if there is any Ada code anywhere that does this?

Fabien-Chouteau, 2016-08-26:
I don't think there is...

fustbariclation, 2016-08-26:
OK.

In C, the bit to set for DHCP is:

#define LWIP_DHCP

Defined in the file:
https://github.com/esp8266/Arduino/blob/master/tools/sdk/lwip/include/lwipopts.h

It is described here: http://lwip.wikia.com/wiki/DHCP

The ST32 documentation (
http://www.st.com/content/ccc/resource/technical/document/application_note/66/dd/f9/3f/5f/75/49/c8/CD00255062.pdf/files/CD00255062.pdf/jcr:content/translations/en.CD00255062.pdf
)

says that you set the MAC address by:

"

When the Server/Client example is used, and in the case of a client,
the firmware sets a different MAC address by replacing the sixth byte
by the CLIENTMAC6 defined in the netconf.c file.

"

netconf.c is here:
https://github.com/mikeferguson/stm32/blob/master/projects/smaldog/netconf.c

My question is how to translate these to Ada.

Not everything, just these two steps.

It's essentially this code ( from here:
https://github.com/mikeferguson/stm32/blob/master/projects/smaldog/netconf.c
)

void LwIP_Init(void)
{
struct ip_addr ipaddr;
struct ip_addr netmask;
struct ip_addr gw;
uint8_t macaddress[6]={0,0,0,0,0,1};
/* Initializes the dynamic memory heap defined by MEM_SIZE.*/
mem_init();
/* Initializes the memory pools defined by MEMP_NUM_x.*/
memp_init();
IP4_ADDR(&ipaddr, 192, 168, 0, 42);
IP4_ADDR(&netmask, 255, 255, 255, 0);
IP4_ADDR(&gw, 192, 168, 0, 1);
//Set_MAC_Address(macaddress);
/* - netif_add(struct netif *netif, struct ip_addr *ipaddr,
struct ip_addr *netmask, struct ip_addr *gw,
void *state, err_t (* init)(struct netif *netif),
err_t (* input)(struct pbuf *p, struct netif *netif))
Adds your network interface to the netif_list. Allocate a struct
netif and pass a pointer to this structure as the first argument.
Give pointers to cleared ip_addr structures when using DHCP,
or fill them with sane numbers otherwise. The state pointer may be NULL.
The init function pointer must point to a initialization function for
your ethernet netif interface. The following code illustrates it's use.*/
netif_add(&netif, &ipaddr, &netmask, &gw, NULL, &ethernetif_init,
&ethernet_input);
/* Registers the default network interface.*/
netif_set_default(&netif);
/* When the netif is fully configured this function must be called.*/
netif_set_up(&netif);
}

Fabien-Chouteau, 2016-08-26:
It looks like there's a lot a code involved, netconf.c is just the tip of the iceberg. You will have to implement this in Ada or maybe write a binding to the C code...

Kensan, 2016-08-31:
Related to the DHCP question: we implemented a DHCP client in Ada see [1]. Maybe the code can be helpful and give some ideas.

[1] - https://www.codelabs.ch/adhcp/

Fabien-Chouteau, 2016-08-31:
Thanks for the link Adrian, it would be interesting to see if this client can run a micro-controller.

fustbariclation, 2016-09-02:
Thank you for that example.

Unfortunately, the problem is that it's not embedded, and it does DHCP
explicitly.

The chip on the STM32 board implements DHCP, as a client, itself. All
that's necessary is to set the MAC address, and set the bit telling it to
use DHCP instead of a fixed IP address, and it does all the rest.

fustbariclation, 2016-09-05:
I read that you'd reshuffled things to neaten it up.

I was wondering what happened to the .gpr files in the 'examples'
directory, like this one:

stm32f7disco_examples.gpr

I found it very useful to be able to build all the demos in one go, as
a sanity check.

Also, just as a thought, how difficult would it be to build the demos
into one integrated demo, with each a sub-task? I think such a
demo-of-demos would be most instructive to newcomers.

Fabien-Chouteau, 2016-09-05:
> I was wondering what happened to the .gpr files in the 'examples' directory, like this one:

Because of the re-organization of projects and the dependency on the `embbeded-runtimes` repository, this is not possible anymore.

> I found it very useful to be able to build all the demos in one go, as a sanity check. 

I miss that too, but we think that the gain is greater than the loss. 

> Also, just as a thought, how difficult would it be to build the demos into one integrated demo, with each a sub-task? I think such a demo-of-demos would be most instructive to newcomers.

I'm in favor of this, if it's done right. I personally think that we need only one example per board, an example that would show as many features as possible in steps:

 - Step 1: LCD
 - Step 2: Touch screen
 - Step 3: Audio
 - Step 4: USB
 - Step 5: Ethernet
 - Step 6: SDcard
 - etc.

closed 2016-10-19 by Fabien-Chouteau