AdaCore/Ada_Drivers_Library #42: Missing config.gpr file

zdun8, 2016-08-24:
Hi,

The serial-ports f429disco example (Ada_Drivers_Library/examples/serial_ports/serial_ports_f429disco.gpr) gpr file includes a file config.gpr which should be stored in the main library folder, but it's not there. 
It seems to be removed in e3a3ceb352240e5ee30fd733b88ce8b8dbdbe098. 
Project doesn't compile. I'm not sure if this is an issue of the example gpr file or sth else. 

Kind regards,
zdun8

plestedr, 2016-08-25:
Similar issue here - The blinky_f4disco.gpr file with'd some other gprs that appear to have been moved. Took a bit of detective work to locate them.

Here's a diff of the first two lines of that project file... I'm using the stm32f407 proc, as you can see.

@@ -1,13 +1,74 @@
-with "../../config.gpr";
-with "../../boards/stm32f4_discovery.gpr";
+with "../../boards/stm32f407_discovery/config.gpr";
+with "../../boards/stm32f407_discovery.gpr";

With that addressed, I ran into another problem - incorrect line terminators - which is a different thread.

Bob

lambourg, 2016-08-25:
This should now be fixed by d455c640d2399a5a2e674001dc604a846ed030b2
Thanks for reporting this issue :)

closed 2016-08-25 by lambourg

plestedr, 2016-08-27:
That did fix my issue. Thank you!