AdaCore/Ada_Drivers_Library #43: 1:79 (style) incorrect line terminator

plestedr, 2016-08-25:
I have cloned the Ada_Drivers_Library and the embedded-runtimes below that.

Trying the build the blinky_f4disco project I initially found that the two with'ed projects were out of place, I fixed that but now I'm getting the line terminator issue on most of the adb and ads files in the embedded-runtimes project. Wondering if I goofed something up when I did that.

I can't fix the the terminators, neither by editing the files by hand nor commanding git to append crlf to every line (I'm on a Win 10 machine). 

Oddly enough, I opened a few of the cases in gps outside of their projects - opened the source code files directly - and they got through the syntax check that way. I even added a comment to make sure it really was the same file.

This issue seems to be halting compilation. There has to be a way to override that but I can't find it.

Toolchain works using the example that was bundled with the compiler, so its not that.

Gotta be some newbie issue that I'm overlooking, but its been days!

I'd like to commit the blinky_f4disco.gpr but I won't until I see it work all the way through.

Thanks in Advance!
Bob

Fabien-Chouteau, 2016-08-25:
Hi Bob,

Did you use GitHub's windows client?

plestedr, 2016-08-26:
Hi Fabien - 

I'm trying both the windows GUI and command line options, I just downloaded the latest from git-scm.com and I get the same result.

When I compile those problem files outside of their projects - load them alone - I don't have problem. When I attempt to fix the problem via editing, I can't change the behavior. Something is happening in the .gpr files; I don't think its the actual .ads or .adb files themselves.

I will keep searching but any advice is welcome!

Bob

plestedr, 2016-08-26:
But no, I did not find a Windows client on Github. Is it server based? 

I'm cloning the repositories, not pulling them.

Bob

Fabien-Chouteau, 2016-08-26:
Please try this configuration:
`git config --global core.autocrlf false`
and re-clone the repositories. It says git not to try to be smart with line termination :)

plestedr, 2016-08-27:
Fabien -

That was the key! I figured that I needed the opposite effect, but there you go. 

Thanks for your help, I'm still learning git et al. When I get a bit smarter I'll create a .gitattributes for the embedded RTs.

I will close this out. 

Bob

closed 2016-08-27 by plestedr

Fabien-Chouteau, 2016-08-27:
You are welcome! I struggled with this more than once :)

Thank you for the .gitattribures hint, we were looking for solution like this :+1: