AdaCore/Ada_Drivers_Library #44: Build stuck while compiling start_rom.S

NitinBhaskar, 2016-08-25:
I tried building "wolf" example for stm32f429 discovery with GPS obtained from gnat-gpl-2016-arm-elf-windows-bin.exe. My build gets stuck while compiling start_rom.S

Fabien-Chouteau, 2016-08-25:
Hi,

Do you have any error message?

NitinBhaskar, 2016-08-25:
No error message. Build gets stuck at `arm-eabi-gcc -c -g -O0 start-rom.S`

NitinBhaskar, 2016-08-25:
I can see following in 'Messages' when I open GPS,

`[2016-08-25 18:56:10] C:\GNAT\2016\share\doc\gprbuild\html\gps_index.xml:1:0: No root element specified
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\GNAT\2016\share\gps\support\ui\icons.py", line 10, in <module>
    from GPS import Icon
ImportError: cannot import name Icon
Welcome to GPS 2016 (20160515) hosted on i686-pc-mingw32
the GNAT Programming Studio
(c) 2001-2016 AdaCore

[2016-08-25 18:56:11] Could not locate exec gnatlswarning: Python module 'board_support' defines class 'BoardLoader' with a method 'gps_started': this is no longer supported: instead, you should override 'setup'`

lambourg, 2016-08-25:
The messages you see in GPS are in fact warnings and can be ignored.

Have  you tried building the example from the command line?

`gprbuild -P wolf_stm32f429disco.gpr -p -XRTS=ravenscar-sfp -XBUILD=Debug -j0`

NitinBhaskar, 2016-08-26:
Yes, I have even tried compiling single file, but no success.. Looks like this is an issue with arm-eabi-gcc. I tried GPS on Linux, there build was successful. I would still prefer working on win7 though.

lambourg, 2016-08-26:
Maybe you can try re-installing the compiler: I just tried on a windows 10 machine and everything seems to work just as expected...

Fabien-Chouteau, 2016-08-26:
And I'm using GNAT GPL 2016 on Win7 (32bit) without any problem so it must be a setup issue.

NitinBhaskar, 2016-08-26:
I have reinstalled on my win7(x64) machine more than 5 times.

NitinBhaskar, 2016-08-26:
Tried with another win7(x64) machine. Build got through. Must be an issue with my machine.

closed 2016-08-26 by NitinBhaskar

Fabien-Chouteau, 2016-08-26:
OK thanks. Happy hacking!