AdaCore/Ada_Drivers_Library #50: stm32f46_79x/stm32_svd-ethernet.ads: definitions for MAC addresses

fustbariclation, 2016-09-05:
I'm a bit confused by this. I'd be grateful for any pointers. The file stm32_svd-ethernet.ads defines the MAC address in two halves. For example:

MACA0HR   at 64 range 0 .. 31;
MACA0LR   at 68 range 0 .. 31;
MACA1HR   at 72 range 0 .. 31;
MACA1LR   at 76 range 0 .. 31;
MACA2HR   at 80 range 0 .. 31;
MACA2LR   at 84 range 0 .. 31;
MACA3HR   at 88 range 0 .. 31;
MACA3LR   at 92 range 0 .. 31;

So each is a 32 bit word. 

They are later set as, for example:

MAC2AH         : MACA2HR_MAC2AH_Field := 16#FFFF#;

The problem I've got is that a MAC address is of the form: 

hh:hh:hh:ll:ll:ll

eg:

01:23:45:67:89:ab

In other words, it's made up of six bytes. The definition in the file is, of course, for four bytes, two sixteen bit words.

Shouldn't the correct form be, for example:

MAC0AH         : MACA0HR_MAC0AH_Field := 32#FFFFFF#;
MAC0AL         : MACA0LR_MAC0AL_Field := 32#FFFFFF#;

So that, for the above example, it'd be set as:

MAC0AH         : MACA0HR_MAC0AH_Field := 32#012345#;
MAC0AL         : MACA0LR_MAC0AL_Field := 32#6789ab#;

fustbariclation, 2016-09-05:
I should probably say that I'd be equally happy, at this stage, if I could set a fixed MAC and IP address. I don't need to use DHCP.

Fabien-Chouteau, 2016-09-05:
Look closer at the definition, MACxHR are not only 32 bits words, they are record with 

```
   subtype MACA0HR_MACA0H_Field is HAL.Short;

   --  Ethernet MAC address 0 high register
   type MACA0HR_Register is record
      --  MAC address0 high
      MACA0H         : MACA0HR_MACA0H_Field := 16#FFFF#;
      --  unspecified
      Reserved_16_30 : HAL.UInt15 := 16#10#;
      --  Read-only. Always 1
      MO             : Boolean := False;
   end record
```
So MACA0LR (4 bytes) plus MAC0HR.MACA0H (2 bytes) give you 6 bytes.

The definition of theses registers looks good, so I'm closing this issue.

Regards,

closed 2016-09-05 by Fabien-Chouteau

fustbariclation, 2016-09-05:
Thank you - that makes sense!