AdaCore/Ada_Drivers_Library #51: stm32-eth.ads - design - expected calling sequences

fustbariclation, 2016-09-05:
Most of the internal buffer variables in stm32-eth.adb are hidden. So I've been wondering what the design intention has been for reading and writing from and to the ethernet card.

For example, in the demo, there is the line:

STM32.Eth.Wait_Packet;

When this completes, it means that a packet has arrived. It is in the buffer Last_Rx, but this is hidden.

If you wanted to get the buffer, what would be the correct method here? If the buffer was visible, then it could be:

STM32.Eth.Wait_Packet.Last_Rx;

But, obviously, that is not the expected method!

Similarly, you define:

Tx_Desc_Type

In STM32.Eth - but there is no procedure linked to it. What method would you be expecting the user to use to send a packet?

I know it's also possible to set it up to call a task on interrupt - if this is the preferred method, is the buffer(packet) passed to the task - or is it passed a pointer to the common buffer?

Fabien-Chouteau, 2016-09-06:
It's part of design that the STM32.<something> packages provides an higher level interface.

Again the STM32.Eth is quite old and maybe not up to the standard of the other drivers. I encourage you to have a look others packages (STM32.I2C, STM32.SPI, STM32.PWM) to see what kind of interfaces we usually expect. 

I also suggest that you start thinking about an HAL.Ethernet interface, again you can look at the other packages in HAL. This should define Ethernet interface that would work on any Ethernet controller. By doing this work, or only starting to think about it, you will be able to design a better interface for STM32.Eth.

fustbariclation, 2016-09-06:
> It's part of design that the STM32. packages provides an higher level

I have been trying, particularly, to complete a project with Ethernet.

Is there a HAL.Ethernet interface anywhere - even an old one?

Do you know anybody who has successfully sent or received a packet with
STM32.Eth? Even if it's terrible code and very old, it'd still be very
useful.

Fabien-Chouteau, 2016-09-06:
> I have been trying, particularly, to complete a project with Ethernet. 

We appreciate your effort, Ethernet is not easy and we are mostly starting from scratch here :)

> Is there a HAL.Ethernet interface anywhere - even an old one? 

Not in Ada, it's on the TODO list. You can have look at ARM CMSIS C interfaces: http://www.keil.com/pack/doc/CMSIS/Driver/html/group__eth__interface__gr.html

>Do you know anybody who has successfully sent or received a packet with STM32.Eth? Even if it's terrible code and very old, it'd still be very useful.

@gingold-adacore did, and all his code is in STM32.Eth (I think).

fustbariclation, 2016-09-09:
> interface__gr.html

Thank you for the reference - I've written to Tristan, and the e-mail
didn't bounce back, but I've not yet had a reply. I know it's holiday
season in Europe, but I thought I should check, is Tristan's adacore e-mail
address still the correct one to use?

stcarrez, 2016-09-27:
I played a little bit with STM32.Eth and it was a nice starting point to understand the Ethernet driver.

The Ethernet driver must have a transmit queue and a receive queue so that you can manage
the TX and RX Ethernet rings. There are many strategies for that and I've implemented a driver
that allows to avoid copies when a frame is sent but also when it is received.

As for now, the Ethernet driver receive queue works very well so that you can recieve network
packets and do whatever you like with them. The transmit queue works but it stops after sending
the first packet but it's a matter of time to fix it.

Check out:
https://github.com/stcarrez/ada-enet

and look at src/stm32/net-interfaces-stm32.adb for the TX and RX ring setup with the transmit
and receive queues. 

I've picked the STM32.Eth and stripped down to the minimal subset for my own needs (MII setup).

Fabien-Chouteau, 2016-09-27:
Thanks for sharing the info! That's an awesome project you have here!

May I suggest that you record a short video of the project in action and put it on YouTube or similar.

fustbariclation, 2016-09-28:
Dear Stephane,

Thank you very much - that's really helpful.

Regards,
Peter

stcarrez, 2016-09-30:
Here is a short video that shows EtherScope in action: [EtherScope an Ethernet Traffic Monitor](https://youtu.be/zEtA-S5jvfY)

closed 2016-10-19 by Fabien-Chouteau