AdaCore/Ada_Drivers_Library #55: examples/filesystem compiles but crashes when run

plestedr, 2016-09-13:
Hi Fabien

Do you have any instructions to go with the filesystem demo? I was able to build it for the stm32f407 and load it on the discovery board, but when I run the program it stops at line 84 in semihosting.adb, at the beginning of the assembler code in Generic_SH_Call. I get this...

Program received signal SIGTRAP, Trace/breakpoint trap.
0x08003176 in semihosting.generic_sh_call (r0=1, r1=536904472) at C:\users\bob\desktop\ada_drivers_library\ARM\cortex_m\src\semihosting.adb:84

I see there's a file called disk_8_partitions.img that comes with the code, do I need to do something with that file?

I get similar results using both the _sfp and _full builds

Thanks!
Bob

Fabien-Chouteau, 2016-09-13:
Hi Bob, 

Are you using st-util?

plestedr, 2016-09-13:
I am using st-util. I don't see a command line option to get st-util to give me its version number, but it should be pretty new.

I was able to compile, load, and run the accelerometer demo, and both of the hello_world demos. Using those some procedures on the filesystem demo gives me the results above. 

Do I need to put that disk_8_partitions.img file anywhere special?

The board is an stm32F407G discovery, the numbers in the upper right corner say M8997D.

TNX
Bob

Fabien-Chouteau, 2016-09-13:
This example requires semi hosting support from st-util which I added to the official repo a few weeks ago. In other words, you need a fresh version of st-util and you need to start st-util with the --semihosting option.

I'm sorry this should be documented. 

Also, for the example to succeed, you have to copy the disk_8_partitions.img in /tmp/.

We can keep the issue open until this is properly documented. 

Regards,

plestedr, 2016-09-14:
Thanks for the tips. I got my st-util executable from st.com. I found your code updates, I download the code and see if I can build a windows .exe unless there's already one out there. I didn't see one.

With that in mind, I have a machine running Fedora 24. The native linux GPS works fine but the ARM version wouldn't run. Different thread, but have you heard anyone else report that?

tnx
Bob

Fabien-Chouteau, 2016-09-14:
I don't remember any problem on Linux. Do you have separate installation directories for native and arm ?

plestedr, 2016-09-15:
I wiped off all of the gps stuff and reinstalled the 64 bit 2016 linux version of the arm-elf. When I run gps I get...
[bob@localhost lib]$ gps
/usr/gnat/bin/gps: /usr/gnat/bin/gps_exe: /lib/ld-linux.so.2: bad ELF interpreter: No such file or directory

For completeness...
[bob@localhost lib]$ uname -a
Linux localhost.localdomain 4.7.2-201.fc24.x86_64 #1 SMP Fri Aug 26 15:58:40 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

Same result using the gnat 2015 arm-elf and 2014 as well, so I think I have an issue with Fedora. 

Couldn't get st-util to build on Windows, so I'll start an issue with Texane and open beer!

Fabien-Chouteau, 2016-09-15:
There is no 64bit release of GNAT GPL for arm.  This error is because gnat arm-elf is a 32bit executable and it cannot find its dependencies in 32bit. I think that's the problem, have a look at your distrib documentation/forums to know how to run 32bit executables.

Regards,

plestedr, 2016-09-15:
Yes, that did it. GPS runs now, however stm32f407_discovery.gpr fails with an unknown attribute "runtime" message. I think I'll put a 32 bit Fedora on a virtual_box and see if I can get that to work.

Bob

Fabien-Chouteau, 2016-09-15:
Can you show me the "runtime" error message ? 

Did you clone with the --recursive option as shown in at the beginning of the README ?

plestedr, 2016-09-15:
I did to the --recursive option. I wiped it out and reloaded... Here's what I did this morning. Also, I only have the arm-elf environment available. The native linux is not installed.

plestedr, 2016-09-16:
Fabien - 

I got everything to work on a native 32 bit Fedora 24 environment running under an Oracle Virtualbox. I don't know why I'm having problems with the embedded runtimes under the 64 bit environment, but they all loaded and compiled just fine in 32 bits.

The filesystem demo ran with your instructions from above. I did clone the stlink package and built it. The demo ends a bit ungracefully, but I think that is by design. 

I'll see if I can document all of this and maybe we can close out the issue. 

Bob

Fabien-Chouteau, 2016-09-16:
Thanks Bob!

Yeah the example is not very friendly :)
It's more a test for a few features I added. I hope that we will be able to add the FATfs driver at some point.

closed 2016-09-17 by plestedr