AdaCore/Ada_Drivers_Library #61: mpu9250 self test configuration

simonjwright, 2016-09-27:
In components/src/motion/mpu9250/mpu9250.adb:193 there is
```
      MPU9250_Write_Byte_At_Register
        (Device, MPU9250_RA_GYRO_CONFIG, Shift_Left (1, FS));
[...]
      MPU9250_Write_Byte_At_Register
        (Device, MPU9250_RA_ACCEL_CONFIG, Shift_Left (1, FS));
```
which is equivalent to [kriswiner’s code](https://github.com/kriswiner/MPU-9250/blob/master/MPU9250BasicAHRS.ino):928 and has, I believe, the [same issue](https://github.com/kriswiner/MPU-9250/issues/78); in both cases it should be `Shift_Left (FS, 3)`.

lambourg, 2016-09-28:
Right. If I'm not missunderstood, this bug has no effect, right? FS being 0.

I'll modify this for at least readability and in case FS is moved to a different value (in which case this would become a real bug).

simonjwright, 2016-09-28:
The current code shifts 1 FS places left, i.e. it sets bit 0. In GYRO_CONFIG this is bit 0 of the Fchoice_b field, which means that FCHOICE in the table below is 2#10#, which means that the bandwidth is 8800 Hz instead of the 250 Hz selected in the line above .. 
[image]

Similarly for ACCEL_CONFIG, which sets bit 0 in the reserved field [2:0] rather than 0 in the ACCEL_FS_SEL field.
[image]

lambourg, 2016-09-28:
Hmmm, very right indeed! Hooops.

lambourg, 2016-09-28:
That's what happens when using shift operations instead of record representation clauses...

simonjwright, 2016-09-29:
... or if you’re translating C which happens to be wrong! (the C has now been fixed too)

lambourg, 2016-09-29:
This is now merged on master. Thanks for having spotted this!

closed 2016-09-29 by lambourg