AdaCore/Ada_Drivers_Library #70: Stm32f103

Jamshoot66, 2016-10-19:
Is it possible to use ada with 103?
How can i get runtime for it?

Fabien-Chouteau, 2016-10-19:
Hi!

> Is it possible to use ada with 103?

I guess you are talking about the STM32F103 which is an ARM Cortex-M3. Then the answer is yes :)

> How can i get runtime for it?

There is a Zero Foot Print (ZFP) run-time for TI's LM3S cortex-M3 core in GNAT GPL 2016: `<GNAT_INSTALL_DIR>/arm-eabi/lib/gnat/zfp-lm3s`

ZFP means there is no tasking support and very few of the run-time Ada features, you can learn more about the run-time profiles here: [Runtime Profiles](http://www.adacore.com/gnatpro/toolsuite/runtimes/)

With a little bit of customization you will be able to create a `zfp-stm32f103` from `zfp-lm3s` and run Ada on the STM32F103. You should only have to touch the files in the `arch/` directory.

Hope this helps,

closed 2016-10-19 by Fabien-Chouteau

huegene, 2016-11-29:
can i post my initial port of the ravenscar-sfp library here? I mean am I allowed to do that?

Fabien-Chouteau, 2016-11-30:
Hi @huegene,

For the run-time it would be in this repository: https://github.com/AdaCore/embedded-runtimes

Note that we at AdaCore are in the process of defining the contributions workflow and rules so we won't be able to accept the contributions just right now.

However you can always fork the repository, work on it, push commits and when everything is ready (on or side and your side) submit the changes to the main repo.

Regards,

Fabien-Chouteau, 2016-12-05:
Hi @huegene  , we are still working on the contribution procedures but if you send me an email at chouteau@adacore.com I will be able to answer some of your questions.