AdaCore/Ada_Drivers_Library #93: New To this

embedThat, 2016-11-25:
I have been working on and off adding features to the Ada_Drivers_Library for a while now on my own, but never committed. I am new to open source and have some questions about how how the copy right, credit is done, also about derivative works and how that all works. If I were to take the STM32 library written in c and port it to Ada. But not a straight over port most of the names are changed and the data structures are completely different but the code follows very similarly. Could I legally add it to the project? I have read the STM32 ULA and I am not sure if it is ok. It talks about derivative works...  

I noticed that most of the work for this project is by AdaCore employees. When adding significant amounts of code to your project how does this work when it comes to credit, because at that point you didn't create the files. (I am new to open source). So many of the files say Copyright (C)  AdaCore what would my files say?

My code hasn't been tested yet but I thought that I should get some of the questions answered and get connected. (I have a few thousand lines written already.)

Fabien-Chouteau, 2016-11-25:
Hi @embedThat, thanks for your message. Now I'm curious to see what you implemented :)

We at AdaCore are in the process of defining the contributions workflow and rules so there's not much I can say as of today.

Stay tuned, I'll make sure to give you the info when they are available.

Fabien-Chouteau, 2016-12-05:
Hi @embedThat , we are still working on the contribution procedures but if you send me an email at chouteau@adacore.com I will be able to answer some of your questions.

Fabien-Chouteau, 2017-06-09:
@embedThat sorry for not keeping you up to date. We published our [contribution guide lines](https://github.com/AdaCore/contributing-howto).

closed 2017-06-09 by Fabien-Chouteau