AdaCore/Ada_Drivers_Library #113: Installation issue?

rveenker, 2017-01-16:
When I try to build the examples and open the project by double-clicking the gpr file, I get 'Could not locate exec gnatls'. Also the project is not loaded and I get the 'default' project instead.

Opening the gpr file from a running GPS does open a project and allows me to build it as well, but I cannot access the Ada runtime library from the help menu. I get empty files instead!

To me this is strange because I do not see this with the Certyflie project (cf_ada_spark.gpr).

I am using the crazyflie virtual machine (Ubuntu 14.0) and the Adacore gnat-gpl-2016-arm-elf-linux compiler. Since I have a STM32f429 discovery board, I tested examples like draw_stm32f429disco.gpr and blinky_f429disco.gpr
Note that from the terminal, the command 'arm-eabi-gnatls' does work.

Any help is appreciated :-)

pmderodat, 2017-01-18:
Hello @rveenker 

This error message “Could not locate gnatls” is definitely unexpected. It looks like GPS ignores the Target project attributes, which could mean that it’s an old GPS release. Can you check you are using the latest release? You can do it with the following command (and here’s the expected output):
```sh
$ gps --version
GPS 2016 (20160515) hosted on x86_64-pc-linux-gnu
```
[cf_ada_spark.gpr](https://github.com/AdaCore/Certyflie/blob/ravenscar-cf-stable/cf_ada_spark.gpr#L53) uses the old way to work with cross toolchains, so it’s expected that an old GPS works with it and not with “new” GPR features such as Target attributes.

Thank you in advance,

rveenker, 2017-01-18:
Yes I can confirm that my GPS is the latest release. I just got this response:
$ gps --version
GPS 2016 (20160515) hosted on i686-pc-linux-gnu

But what I find strange is that the certyflie project does not suffer from this.

rveenker, 2017-01-18:
Just found that when I remove the project files that I don't need from the directory and leave for instance just draw_stm32f429disco.gpr, I can open the project from the file manager by double-clicking.
Also the runtime help (show the Ada runtime packages) seems to work now although I still get the gnatls message.

pmderodat, 2017-01-18:
Ok, thank you for the extra details. As I don’t use a file manager myself, I tried to run GPS with the following command to immitate it:
```shell
$ gps draw_stm32f429disco.gpr
```
As a result, GPS indeed opens the GPR file as a *source file* and loads the default project. I don’t get the `gnatls` error message, though, but I assume that’s because I also installed GNAT for `x86_64-linux` (which provides a `gnatls` program). Anyway I think that’s kind of expected for GPS: in order to load project `X.gpr`, one must run:
```shell
$ gps -P X.gpr
```
… which correctly loads X.gpr as a *project* and which properly sets the `arm-eabi` target.

I think I understand why you don’t have this problem with the Certyflie project file: when launched without a `-P` argument, GPS will automatically load the project file in the current directory *only* if there is exactly one project file there. That also explains why removing all project files you don’t need solves your issue.

Anyway, project files in this repository are fine: it’s rather a GPS issue. :-)

closed 2017-01-18 by pmderodat

Joebeazelman, 2022-02-27:
I'm getting the same issue trying to run install_dependencies.py on MacOS.

Joebeazelman, 2022-02-27:
I am getting the same issue when I try to run install_dependencies.py:

Traceback (most recent call last):
  File "/Users/bob/Projects/Repos/NXP_LPC1768/Ada_Drivers_Library/repo/scripts/./install_dependencies.py", line 77, in <module>
    os.path.join(os.path.dirname(shutil.which("arm-eabi-gnatls")),
  File "/usr/local/Cellar/python@3.9/3.9.10/Frameworks/Python.framework/Versions/3.9/lib/python3.9/posixpath.py", line 152, in dirname
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType

Fabien-Chouteau, 2022-02-28:
@Joebeazelman do you have `arm-eabi-gnatls` in your PATH?

Joebeazelman, 2022-03-03:
Probably not.  I use ALR to manage my toolchain for me. I just use ALR toolchain --select. Does ALR require the project folder to have some sort of initialization?

Fabien-Chouteau, 2022-03-09:
This Ada_Drivers_Library repository is not made to be use with Alire yet.
You have to set a compiler in path before you run the script, that's why it is failing.