AdaCore/Ada_Drivers_Library #163: need svd source for CRC unit to have workaround

pat-rogers, 2017-02-25:
We lost the work-around for the compiler bug Q207-001 in stm32_svd-crc.ads, in which a register component has an extra "with Volatile" even though that is already specified at the record level.  I had manually added that to the Ada file but I don't know how to specify that in the SVD input part.

Fabien-Chouteau, 2017-02-27:
This is tricky. Maybe you can comment the lines that generate STM32 SVD code in svd.mk, and with a comment explain that until the compiler is fixed we should be extra careful with STM32 SVDs.

pat-rogers, 2017-02-28:
On 2/27/2017 1:06 PM, Fabien Chouteau wrote:
> This is tricky. Maybe you can comment the lines that generate STM32 SVD code in svd.mk, and with a comment explain that until the compiler is fixed we should be extra careful with STM32 SVDs.

Done but not yet checked in.

Fabien-Chouteau, 2017-06-09:
@pat-rogers I think we can close this one, right?

pat-rogers, 2017-06-09:
On 6/9/2017 11:44 AM, Fabien Chouteau wrote:
> @pat-rogers I think we can close this one, right?

Yes, thanks.

closed 2018-06-01 by Fabien-Chouteau