AdaCore/Ada_Drivers_Library #173: I2C w/16bit mem addr

morbos, 2017-03-07:
I see a punch through issue wrt successive .DR.DR writes for a 16bit mem addr.

In stm32-i2c.adb:621 adding this wait after the end case seems to clean up the issue. Should be benign for 8bit mem addr also. I have not looked at the other 16bit handlers but I assume they will also need this treatment. Certainly read needs it. My sensor is an ST VL6180X Range and ALS sensor. It uses 16bit addresses. On the logic analyzer I do see the address being stepped on by the following read command thereby culling the DR of the LSB for the parts mem addr.

      Wait_Flag (This, Tx_Data_Register_Empty, False, Timeout, Status);

      if Status /= HAL.I2C.Ok then
	 return;
      end if;
      
      --  We now need to reset and send the slave address in read mode
      This.Periph.CR1.START := True;

      Wait_Flag (This, Start_Bit, False, Timeout, Status);

Fabien-Chouteau, 2017-03-07:
Hi @morbos ,

It's possible that we never tested the 16bit mem address mode, so I would not be surprised to see a problem here. 

> In stm32-i2c.adb:621 adding this wait after the end case seems to clean up the issue. Should be benign for 8bit mem addr also. I have not looked at the other 16bit handlers but I assume they will also need this treatment. Certainly read needs it. 

It looks OK. Can you send a pull request with that change? If it doesn't break the other I2C mode we will be happy to have it.

> My sensor is an ST VL6180X Range and ALS sensor. 

Interesting, my colleague @lambourg is working on the VL53l0X. Have a look at this branch: https://github.com/AdaCore/Ada_Drivers_Library/tree/vl53l0x

Thanks,

morbos, 2017-03-10:
Hi Fabien, @lambourg's work is very helpful indeed. We have the same 4 char display on the shield. The vl5310x needs only an 8bit addr. One beef I have with the vl6180x is there are 30 undocumented writes (ex: write (this.Port, 16#0097#, 16#fd#); etc)  to get the device to do useful work. After Pat updated the BNO055 example to use the generic, I wanted to try my hand at a sensor as a generic, thus the vl6180x (via a shield) on an F4 Nucleo board (STM32F401RE).

Hedley