AdaCore/Ada_Drivers_Library #195: STM32L4xx

morbos, 2017-05-21:
Before a wheel reinvention... anyone had a stab at Ada_Drivers_Library for the L4 series. Sifting through the ST F40x -> L4 porting docs, i2c for ex:  (that statement below is no joke... the svd file is quite diff).

_Register configuration is very different in STM32F405/415 line and STM32F407/417 line vs STM32L4 Series. Please refer to STM32L4 Series reference manuals for details._

My thought would be minimally to make a new drivers dir driversL4 or some such.

Its a bare minimum target. 128k flash, 64k sram. Prob sfp only.

Fabien-Chouteau, 2017-05-22:
Hi Hedley,

So you already have the SVD files? Looking at the documentation, it seems like the STM32L4 peripherals are closer to the STM32F**7** peripherals... A simple way to know is to try to reuse the STM32F7 drivers with STM32L4 SVD and see if it compiles. If it does there's a good chance that the drivers will work.
If it doesn't you can see how it's done for [STM32F4 vs STM32F7](https://github.com/AdaCore/Ada_Drivers_Library/tree/master/arch/ARM/STM32/drivers) and have specific drivers in a sub directory when needed. 

We can help you with the directory and project file hierarchies.

Thanks for looking at this!

morbos, 2017-05-22:
The zip for all the L4 svd files is on here: http://www.st.com/en/microcontrollers/stm32l432kc.html (look for 'STM32L4 System View Description'). I understand on the dir stuff, thanks for the tip on F7 vs L4. Curr I have a bare minimum elf that gets to the PLL init. Now time to add back the missing bits. There is a cute board I bought for ~$10 (5x1.8cm). Its a Nucleo with a 32 pin(!) 80mhz stm32l432kc on it. Comes with an stlinkV2 too. Openocd 0.10 works fine. It has 256k flash and 64k ram. Enough to run Ada no trouble.

morbos, 2017-05-22:
Fabien, one other Q, how are the i-stm32*.ads files generated? svd.mk handles the arch files but for the embedded-runtimes/bsps? and now, this might expain why the PLL init -> hang(). ;)

Fabien-Chouteau, 2017-05-23:
The i-stm32 packages are in the run-times, they are generated from the same SVD files the difference is the root package used in the SVD2Ada command line:

`./svd2ada -p Interfaces.STM32`

In Setup_PLL, there's a few busy loops waiting for a clocks to stabilize. If you have a debug link, it should be quick to identify.

morbos, 2017-05-30:
Tnx for the tips Fabien. I got a partial lib built for it. Pll locks off of msi 4mhz int clock, gpio is up, timer untested but Ada realtime works, I2C works, spi untested. I have enough working now to put in my make with Ada project using that tiny platform. Once the lib built clean, I had two minor bugs in my app but none in the lib. Found a couple of svd file errors for the part. Wonder how they validate those files...

Fabien-Chouteau, 2017-05-30:
That's awesome!

I will close this issue and wait for your pull requests :wink:

closed 2017-05-30 by Fabien-Chouteau