AdaCore/Ada_Drivers_Library #213: Adding support for STM32F103

minexew, 2017-08-07:
Hi,
I'm interested in adding support for the STM32F103 line of microcontrollers. I would appreciate any guidance on where to start.

What I'm particularly confused about is the relation between Ada_Drivers_Library and AdaCore/embedded-runtimes. To support the line of MCUs, would both projects need to be updated?

minexew, 2017-08-07:
Also, embedded-runtimes vs bb-runtimes?

Fabien-Chouteau, 2017-08-08:
Hi @minexew,

> I'm interested in adding support for the STM32F103 line of microcontrollers. I would appreciate any guidance on where to start.
> 

That's great! We will do our best to help you.

The first thing to do is to have a working run-time. I know there are some discussion about STM32F103 support in the IRC #ada channel, we also have a [porting guide on the blog](http://blog.adacore.com/porting-the-ada-runtime-to-a-new-arm-board) however it was published one and half year ago so part of it is not relevant anymore.

> What I'm particularly confused about is the relation between Ada_Drivers_Library and AdaCore/embedded-runtimes. To support the line of MCUs, would both projects need to be updated?

About Ada_Drivers_library vs bb-runtimes, I wrote a small explanation text that I want to get reviewed before publishing it. I will keep you updated. The short answer is that you will first need a run-time (in the repo bb-runtimes) then you will be able to work on drivers (in the repo Ada_Drivers_Library).

About `bb-runtimes` vs `embedded-runtimes`, `embedded-runtimes` is deprecated. It is an intermediate representation between the sources (now in `bb-runtimes`) and the installation that you get in the compiler. Now that we published the `bb-runtime` repository, there's no need to use `embedded-runtimes` anymore. New contributions should be made to `bb-runtime` directly and we will add a notice in the `embedded-runtimes` repo.

minexew, 2017-08-09:
Sorry for the mistake, I meant F303 which is based on the Cortex M4F..

Fabien-Chouteau, 2017-08-09:
That means it will be easier for you to port the run-time, it will mostly be a BSP work. You can start from here [AdaCore/bb-runtimes/tree/gpl-2017/arm/stm32](https://github.com/AdaCore/bb-runtimes/tree/gpl-2017/arm/stm32), copy the `stm32f40x` directory into a new `stm32f30x` and change what is needed for the 303.

closed 2018-06-01 by Fabien-Chouteau

minexew, 2018-06-01:
Yeah... I didn't get far.