AdaCore/Ada_Drivers_Library #229: Fail to build on STM32F746, STM32F769, STM32F469 but ok on STM32F429

stcarrez, 2017-11-18:
When trying to build the examples with GNAT ARM 2017, I'm not able to build the STM32F746, STM32F769, STM32F469 examples but the STM32F429 is building ok.  The examples are failing
with the following error:

    Error: no compiler found for language 'ada', target = arm-eabi, runtime = ravenscar-sfp-stm32f746disco
    draw_stm32f746disco.gpr:3:09: warning: no compiler specified for language "Ada", ignoring all its sources
    stm32f746_discovery.gpr:11:17: no languages defined for this project
    stm32f746_discovery.gpr:23:25: warning: libraries are not supported on this platform
    stm32f7x.gpr:6:17: no languages defined for this project
    stm32f7x.gpr:8:25: warning: libraries are not supported on this platform
    HAL.gpr:3:17: warning: no compiler specified for language "Ada", ignoring all its sources
    HAL.gpr:7:25: warning: libraries are not supported on this platform
    sdmmc.gpr:4:17: no languages defined for this project
    sdmmc.gpr:6:25: warning: libraries are not supported on this platform
    cortex_m7.gpr:4:17: no languages defined for this project
    cortex_m7.gpr:12:25: warning: libraries are not supported on this platform
    components.gpr:5:17: warning: no compiler specified for language "Ada", ignoring all its sources
    components.gpr:10:25: warning: libraries are not supported on this platform
    middleware.gpr:4:17: warning: no compiler specified for language "Ada", ignoring all its sources
    middleware.gpr:15:25: warning: libraries are not supported on this platform
    ravenscar_support.gpr:4:17: warning: no compiler specified for language "Ada", ignoring all its sources
    ravenscar_support.gpr:9:25: warning: libraries are not supported on this platform
    common.gpr:3:09: warning: no compiler specified for language "Ada", ignoring all its sources
    gprbuild: "draw_stm32f746disco.gpr" processing failed

I've tried to track down the issue by comparing the stm32f746disco.gpr with stm32f429_discovery.gpr and I guess this is related to the Ada runtime, but then I got lost in how the runtime is built.
After changing the line:

     for Runtime ("Ada") use RTS_Profile & "-stm32f746disco";
into

     for Runtime ("Ada") use RTS_Profile & "-stm32f429disco";

It is able to find the Ada compiler.  Of course it does not compile because this is not the correct runtime.
All this with a fresh checkout of the Ada_Drivers_Library and a run of ./scripts/install_dependencies.py.
The embedded-runtimes/BSPs seems to contain all the required files.

Any idea what is going wrong or how to investigate this issue?

stcarrez, 2017-11-18:
I've re-installed the compiler and it is working now.

Sorry for the issue.  Still, it's not easy to track this kind of issue!

stcarrez, 2017-11-19:
Guys, I realize that the GNAT runtime (base_runtimes) which was part of Ada_Drivers_Library/embedded-runtimes in 2016 has been removed by https://github.com/AdaCore/embedded-runtimes/commit/61af72d90582e88229c33ba670401bd736814995. I missed that when I switched to GNAT 2017.

Looking at the sources of MakeWithAda 2017 competitors, it seems that most of them are in the same situation as me.  For example, German Rivera has made some nice porting on NXP Kinetis K64F
https://github.com/jgrivera67/embedded-runtimes/commit/e642eda979b044ee75c5bf0c805805b981b8aea0 and he had to change several files in the GNAT runtime.  It will be hard for him to switch to the latest Ada_Drivers_Library.

Do you have recommendations or suggestions on how we could customize the GNAT runtime in the future?

Fabien-Chouteau, 2017-11-21:
Hi @stcarrez ,

That's because we changed the way we publish the run-times. The old embedded-runtimes repo was a strange middle ground between the repo that we maintain at AdaCore and what is distributed in the GNAT releases.

We changed that and now publish on GitHub the real repo where we develop and maintain the BSPs: https://github.com/AdaCore/bb-runtimes

For Ada_Drivers_Library, there a script: `scripts/install_dependencies.py` that will clone this repo and install all the required run-times. This is explained in the [getting started](https://github.com/AdaCore/Ada_Drivers_Library/tree/master/examples) doc.

> Looking at the sources of MakeWithAda 2017 competitors, it seems that most of them are in the same situation as me. For example, German Rivera has made some nice porting on NXP Kinetis K64F
> jgrivera67/embedded-runtimes@e642eda and he had to change several files in the GNAT runtime. It will be hard for him to switch to the latest Ada_Drivers_Library.
> 

We understand that our change is disruptive but it was necessary for us to be able to accept contributions on the run-time BSPs. Note that the porting efforts are not lost, the run-times are mostly the same and only the architecture of the repo is different.

> Do you have recommendations or suggestions on how we could customize the GNAT runtime in the future?

So far out best documentation is [here](https://github.com/AdaCore/bb-runtimes/tree/master/doc/porting_runtime_for_cortex_m). Don't hesitate to ask questions on the bb-runtimes GitHub repo.

closed 2017-11-21 by Fabien-Chouteau