AdaCore/Ada_Drivers_Library #240: Attribute failure

Rubber-Duck-999, 2018-01-09:
[Log.txt](https://github.com/AdaCore/Ada_Drivers_Library/files/1615188/Log.txt)

I have tried using one of the demonstration projects in /Ada_Drivers_Library/ARM/STM32/driver_demos/ but I recieve issues with runtime and creat missing dirs attribute. I am using bard board arm elf 2017 to port projects on the stm32f429 discovery board. I am also using windows 10. I would like to be able to build the demonstration projects. Can anyone help with this?

Rubber-Duck-999, 2018-01-10:
Hi @Fabien-Chouteau,

I am using version 2017.

Thanks

Fabien-Chouteau, 2018-01-11:
The error messages seems to indicate an old version of gprbuild. Can you give us the results of this command so we can see the exact version you are using:
`gprbuild --version`

Regards,

closed 2018-06-01 by Fabien-Chouteau