AdaCore/Ada_Drivers_Library #286: Cannot run flasher example in GPS2018 on STM32F4 Discovery Board

zahidzs, 2019-01-19:
I just tried to use the default example Led Flasher in the examples where the GPS 2018 was installed, however, after building and flashing to the board, nothing happens. Console message tells me to do a hard reset but it won't work properly. Here are the compilation messages.

st-flash --reset write C:\GNAT\2018-arm-elf\share\examples\gnat-cross\led_flasher-stm32f4\obj\led_flasher.bin 0x08000000

2019-01-19T18:12:59 INFO src\usb.c: -- exit_dfu_mode
2019-01-19T18:12:59 INFO src\common.c: Loading device parameters....
2019-01-19T18:12:59 INFO src\common.c: Device connected is: F4 device, id 0x10036413
2019-01-19T18:12:59 INFO src\common.c: SRAM size: 0x30000 bytes (192 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 16384 bytes
2019-01-19T18:12:59 INFO src\common.c: Attempting to write 61644 (0xf0cc) bytes to stm32 address: 134217728 (0x8000000)

st-flash 1.3.1
Flash page at addr: 0x08000000 erased

Flash page at addr: 0x08004000 erased

Flash page at addr: 0x08008000 erased

Flash page at addr: 0x0800c000 erasedEraseFlash - Sector:0x0 Size:0x4000 EraseFlash - Sector:0x1 Size:0x4000 EraseFlash - Sector:0x2 Size:0x4000 EraseFlash - Sector:0x3 Size:0x4000 2019-01-19T18:13:01 INFO src\common.c: Finished erasing 4 pages of 16384 (0x4000) bytes
2019-01-19T18:13:01 INFO src\common.c: Starting Flash write for F2/F4/L4
2019-01-19T18:13:01 INFO src\flash_loader.c: Successfully loaded flash loader in sram

2019-01-19T18:13:02 INFO src\common.c: Starting verification of write complete

2019-01-19T18:13:02 INFO src\common.c: Flash written and verified! jolly good!

enabling 32-bit flash writes
size: 32768
size: 28876

[2019-01-19 18:13:02] process terminated successfully, elapsed time: 03.395s

Board works fine since I am able to run a similar demo program on KEIL uvision4, but on this new version of GPS2018, the project just won't run. A couple of years ago, I was able to do it in I think GPS 2016 but not with this one. I have ST-LINK v. 4.3.0, can I program the board by just building the files in GPS and then burn into the board through the ST-LINK?

pat-rogers, 2019-01-19:
On 1/19/2019 9:20 AM, zahidzs wrote:
> I just tried to use the default example Led Flasher in the examples 
> where the GPS 2018 was installed, however, after building and flashing 
> to the board, nothing happens. 
> ...
> Board works fine since I am able to run a similar demo program on KEIL 
> uvision4, but on this new version of GPS2018, the project just won't 
> run. A couple of years ago, I was able to do it in I think GPS 2016 but 
> not with this one. I have ST-LINK v. 4.3.0, can I program the board by 
> just building the files in GPS and then burn into the board through the 
> ST-LINK?

Make sure you're using the expected board, since the LEDs are 
board-specific. This demo used to be set up for the F407 Discovery 
board, but now runs on the F429 Disco boards. (See the README in the 
directory.)

Best regards,

Pat

-- 
Patrick Rogers, PhD
rogers@adacore.com
+1 646 375 0739 office
+1 646 358 3289 fax

zahidzs, 2019-01-19:
I am using the STM32 F4 Discovery with the stm32f407 chip on it. Is there any other way to get the program running? One thing, on the board description, Embedded ST-LINK/V2 is written whereas I am using the 4.3.0 version. Is it because of that?

zahidzs, 2019-01-19:
"led_flasher-stm32f4:  This example targets the STM32F4 series of microcontrollers from STMicroelectronics with a simple LED flashing application."   This is the description written in the readme file. I hope this is the proper example for the stm32f4 discovery board with the stm32f407 chip on it.

Regards, 

Zahid.

Fabien-Chouteau, 2019-01-21:
Hi @zahidzs ,

> This is the description written in the readme file. I hope this is the proper example for the stm32f4 discovery board with the stm32f407 chip on it.

Which README file are you reading? 

The file `2018-arm-elf/share/examples/gnat-cross/led_flasher-stm32f4/README.txt` says: `The example targets the SMT32F429-Discovery board`.

You can find examples for the STM32F407 in this repository: https://github.com/AdaCore/Ada_Drivers_Library/tree/master/examples/STM32F4_DISCO

Regards,

closed 2019-02-18 by Fabien-Chouteau