AdaCore/Ada_Drivers_Library #294: Ravenscar_Fill: plan to support Text_Float_IO

LaetitiaEl, 2019-01-31:
Hi,
Do you have a plan to support the Text_Float_IO (a-tiflio library)?
My need is to convert a Float to a String to send it via UART for telemetry/debug purpose.

Will it work if I add the following files to the bb-runtimes  : (Ada_Drivers_Library/boards/stm32f746_discovery/src/full/) for test and build the board using gprbuild/gprinstall : 
- a-flteio : for Ada.Float_Text_IO
- a-tiflio : for Ada.Text_IO.Float_IO
- a-wtflau : for Ada_Text_IO.Float_Aux
- a-wtgeau
- i-cstrea
- s-fileio
- s-ficobl
- s-param

Add the conditions (libs available in bb-runtimes) (how to add them):
- Add_Image_Float:yes to have s-imgrea (available in bb-runtimes) needed by: a-wtgeau
- Add_Value_Float:yes to have s-valrea (available in bb-runtimes) needed by : a-wtgeau
- Add_Streams:yes to have a-stream (available in bb-runtimes) needed by : s-ficobl

Thank you in advance,

Fabien-Chouteau, 2019-01-31:
Hello @LaetitiaEl,

> Do you have a plan to support the Text_Float_IO (a-tiflio library) in the ravenscar_full?

Which package goes in what run-time, is what we call the run-time profile. There are 3 profiles for the bare metal embedded run-times: ZFP, Ravenscar SFP, Ravenscar Full.

We don't have plans to add those packages in the Ravenscar Full proile. 

> My need is to convert a Float to a String to send it via UART for telemetry/debug purpose.

You should be able to do `'Img` on float types, this will convert the float to `String`. However you don't have control on the representation.

A solution that I used once was to convert to a fixed point type, and then use `'Img`:
```.ada
with Ada.Text_IO; use Ada.Text_IO;
procedure Learn is
   type Fixed_Type_1 is delta 0.00001 digits 18;
   type Fixed_Type_2 is delta 0.001 digits 18;
begin
   Put_Line (Float (123456789.123456789)'Img);
   Put_Line (Fixed_Type_1 (123456789.123456789)'Img);
   Put_Line (Fixed_Type_2 (123456789.123456789)'Img);
end Learn;
```
Output:
```
1.23457E+08
123456789.12345
123456789.123
```

> Will it work if I add the following files to the bb-runtimes : (Ada_Drivers_Library/boards/stm32f746_discovery/src/full/) for test and build the board using gprbuild/gprinstall :

It will work, but it will mean that you are the only one capable of compiling your application. Because it will depend on a non standard run-time profile.

> Thank you in advance,

You are welcome,

closed 2019-01-31 by Fabien-Chouteau

LaetitiaEl, 2019-01-31:
The example works very fine in the STM32H746 Nucleo!
I managed to send the data like following (not proper way) to test:
     for ch of FixedFloat(AccelRates(X))'Img loop
         c := Character(ch);
         data_1B(0) := Character'Pos(c);
         UART_OUT.Transmit(data_1B,status);
      end loop;

Fabien-Chouteau, 2019-02-01:
> Do you have an example how to how to link the Ada.Text_IO.Put_Line to a UART?

It would require a modification of the run-time, and there is no real advantage to do that in your case.