AdaCore/Ada_Drivers_Library #301: pyocd-flashtool is deprecated

gymboh32, 2019-04-16:
I recently bought myself a Micro:bit and decided to try writing my project in Ada, which inevitably led me here.  I figured I would get started with the example scrolling text but when I tried flashing to the board I was presented with the following error messages:
[image]
[image]

I dug a bit into the exception message to see what was going on to find the whole Cortex_M class was unsupported
[image]
 
So I thought maybe updating to the latest pyocd package would solve this only to find the following messages:
[image]
[image]

Debugging to the board works but flashing does not seem possible.  
With these errors I don't see how others were able to flash to the Microbit successfully, and it might be more of a question for the folks over at pyOCD.  
Also, with the discovery of GPS using deprecated tools I was wondering if there was any path to updating them, not that it seems to make a difference in my case.

Any help forward on how to flash the Microbit would be appreciated.

Fabien-Chouteau, 2019-04-16:
Hi @gymboh32,

Which version of pyocd are you using?
I am on 0.19.0 (ubuntu), I see the deprecated message but the tool still works.

Maybe it is an access right problem, are you on Linux?

Regards,

gymboh32, 2019-04-17:
I'm running ubuntu 18.04 LTS
pyocd version: 0.19.0 
2018 GNAT came with pyocd version 0.8.0  

I don't think it is an access problem because I can debug without an issue (on both versions of pyocd).  I had also added the udev rules just to be sure of that.  

What concerns me is the warning about not being able to program the flash.  Which is exactly what I'm trying to do.  Is pyocd misidentifying the target type?

gymboh32, 2019-04-17:
I was able to successfully flash.  
It was the target type being misidentified as a `cortex_m`  when it should be `nrf51`.

Fabien-Chouteau, 2019-04-18:
How did you fix the problem?

gymboh32, 2019-04-19:
I had to run `pyocd flash -t nrf51 /path/to/main.bin` in the terminal.  Using the `flash to board` option in GPS runs `pyocd-flashtool -a 0x00000000 /path/to/main.bin` which automatically identifies the target type as `cortex_m`.  
The micro:bit has a `nrf51822` which is built around the `cortex_m` so it makes sense that the pyocd tool chose the wrong target type.

Is there a way to configure the target type (or any other pyocd switches) in the gpr file or as a preference in GPS?

yago83, 2019-07-11:
Hello,
I had same problem, I think it is a problem of GNAT-GPS, I had posted [here](https://github.com/AdaCore/gps/issues/84#issue-466583560)

Fabien-Chouteau, 2019-07-11:
Hello @yago83 

Does `pyocd flash -t nrf51 /path/to/main.bin` work for you?

yago83, 2019-07-11:
Hello @Fabien-Chouteau 

Thank you, it works perfect!
It seems it is being investigating how to fix this problem in GNAT-GPS

Thank you!

Fabien-Chouteau, 2019-08-20:
@gymboh32 @yago83 Do you mind trying again with GNAT community 2019?

I am on Ubuntu 18.04 and everything works out of the box.

closed 2019-08-20 by Fabien-Chouteau