AdaCore/Ada_Drivers_Library #305: Sipeed RISC-V support

ohenley, 2019-05-30:
Feel free to close if not relevant but do you plan, or is there already support for the RISC-V Sipeed boards? https://www.seeedstudio.com/sipeed

Some friend of mine at PolyMtl told me they work great.

Thanks.

Fabien-Chouteau, 2019-05-31:
Hi @ohenley,

I'm not aware of efforts to support these boards from the community. I have one Sipeed Maix Bit board myself but no time to work on it...

The chip is a Kendryte K210, making a compiler and ZFP run-time shouldn't be too much effort.
On the other hand I cannot find a good documentation off the chip, so driver development may require to reverse engineer the drivers provided by the vendor.

closed 2019-05-31 by Fabien-Chouteau

ohenley, 2019-05-31:
those are too shallow?
https://github.com/kendryte/kendryte-doc-datasheet/tree/master/en
https://github.com/kendryte/kendryte-doc-standalone-programming-guide/tree/master/en

JustAnother1, 2019-05-31:
Yes they are. They only describe the high level specifications of the chip and how to use their Driver APIs. What is needed to implement drivers would be the information about the Special Function Registers. That are memory locations that are used to configure the peripherals. The location and size of these registers must be known. Also the meaning of all the bits in them.

For example an UART usually has a register used to write the data into that shall be send out. Another(or the same) to read the received data from. Another to configure the baud rate, and stuff and another one to signal events. Events could be we received a byte or sending of the byte has finished. The event bits can the usually also be mapped to interrupts or to an DMA. How to do that must be specified. Again this is probably done by setting the correct bits at the correct locations in some SFR.

So yes the linked documentation is to "shallow" to write drivers.

Fabien-Chouteau, 2019-06-04:
Could be worth asking the vendor for more detailed docs, and even the SVD (one can dream :smile: ).

ohenley, 2019-06-04:
I opened an issue. Please step in for any 'technical sharp edge' follow up as I am not an embedded guru ...

Thx

https://github.com/kendryte/kendryte-doc-datasheet/issues/14