AdaCore/Ada_Drivers_Library #310: Missing Atomic instructions for FE310-G000 in Hifive1

jumapico, 2019-09-09:
[The FE310 supports](https://sifive.cdn.prismic.io/sifive%2Ffeb6f967-ff96-418f-9af4-a7f3b7fd1dfc_fe310-g000-ds.pdf) the RV32IMAC ISA options.

Note: I don't add a (trivial) patch due that after add the 'a' (for atomic) in
https://github.com/AdaCore/Ada_Drivers_Library/blob/8677f892a9b24e4a3fb48e34850a17af9ef02b66/scripts/config/boards.py#L155
I could not found the `startup-gen` script to regenerate the files `boards/HiFive1/hifive1_zfp.gpr` and `boards/HiFive1/src/zfp/adl_config.ads`.

Fabien-Chouteau, 2019-09-11:
Hello @jumapico ,

> [The FE310 supports](https://sifive.cdn.prismic.io/sifive%2Ffeb6f967-ff96-418f-9af4-a7f3b7fd1dfc_fe310-g000-ds.pdf) the RV32IMAC ISA options.

This was intended as a "who can do more can do less" solution. Since the compiler will not emit atomic instructions by itself (at least I don't think so) this should be OK.
Did you find a situation where this is a problem?

> I could not found the `startup-gen` script to regenerate the files `boards/HiFive1/hifive1_zfp.gpr` and `boards/HiFive1/src/zfp/adl_config.ads`.

`startup-gen` is a tool that we are working on and is soon to be released on GitHub.

In the general case you don't need to re-generate the board support. Do you want to support another board?

Regards,

jumapico, 2019-09-11:
Thanks for the Fabien explanation.

If the intention of `Runtime_Name_Suffix` is for * what is compatible * instead of * what is the hardware * you are right.

Sorry for the noise.

Regards,

closed 2019-09-11 by jumapico

Fabien-Chouteau, 2019-09-11:
> If the intention of `Runtime_Name_Suffix` is for _what is compatible_ instead of _what is the hardware_ you are right.

Right, this is also part of a new scheme to support more hardware that we did not explain yet because we are still working on it.

> Sorry for the noise.

No problem,