AdaCore/Ada_Drivers_Library #312: Measuring time in micro seconds

pirib, 2019-10-03:
Hi there,

I am programming on BBC micro-bit and trying to get an ultrasonic sensor HC-SR05 to work. In a nutshell one needs to set one of the pins HIGH for 10 Micro seconds, and then read off the length of the pulse from another pin, and from my interpretation, I need to measure time while the pin is set to HIGH. From watching Arduino examples, that time is in Microseconds. The problem is, time defined on microbit is in Milliseconds, meaning my measurement will almost always be 0. 

Could you please help me with this? Thanks!

I am using the following construct to measure time in milliseconds.

`
t_start := MicroBit.Time.Clock;
MicroBit.Time.HAL_Delay.Delay_Microseconds(100);  -- Or say a loop in here.
d := Integer(MicroBit.Time.Clock) - Integer(t_start);
`

Fabien-Chouteau, 2019-10-04:
Hi @pirib,

The implementation of the clock in this micro:bit board support was made to be really simple, so indeed there is only a 1 millisecond granularity.

I see two solutions:
 - Improve the `MicroBit.Time` to be able to provide better time resolution. To do that, the `Clock` function should not only return the `Clock_Ms` global variable, but also look at the current value of the `RTC` counter. But it is not as simple as that, there are some pitfalls when mixing a software and a hardware clock.

 - The other solution is for you to use the other `RTC` to count time.

pirib, 2019-10-06:
Hi Fabien-Chouteau,

Thanks for your respond, I will take a look at using the other RTC. Given that my project is due very soon, I am using this solution (leaving it here so others might find it).

`procedure Main is

    LedPin : constant MicroBit.IOs.Pin_Id := 0;
    EchoPin : constant MicroBit.IOs.Pin_Id := 8;
    TrigPin : constant MicroBit.IOs.Pin_Id := 12;

    d: Integer;

begin

    d := 0;

    MicroBit.IOs.Set(TrigPin, False);
    MicroBit.Time.Delay_Ms(10);

    MicroBit.IOs.Set(TrigPin, True);
    MicroBit.Time.HAL_Delay.Delay_Microseconds(10);
    MicroBit.IOs.Set(TrigPin, False);

    outerloop:
    loop

	if MicroBit.IOs.Set(EchoPin) then

	    innerloop:
	    loop
		MicroBit.Time.HAL_Delay.Delay_Microseconds(1);
		if MicroBit.IOs.Set(EchoPin) then
		    d := d + 1;
		else
		    MicroBit.Display.Display( Integer'Image(  Integer(Float(d)*0.034/2.0) ) );
		    exit outerloop;
		end if;

	    end loop innerloop;

	end if;

    end loop outerloop;

    MicroBit.Time.Delay_Ms(1000);
    MicroBit.IOs.Set(LedPin, False);

end Main;`

Fabien-Chouteau, 2019-10-07:
> In a nutshell, it is a simple loop that measures how long in microseconds a given pin (EchoPin in this case) has been set to HIGH. Not the best way of doing it, but it does its job.

Be careful, the `Delay_Microseconds` might not have the accuracy that you expect.
In this case it is probably better to just do a simple counter without any call to delay.
You can then do some measurements to find the relation between count you get and distance.

> Thanks again for your help. Would be really nice if you guys added more documentation to the library, getting Bluetooth to work so far has been an incredibly painful expereince.

Thank you for the feedback. The Bluetooth support of the micro:bit is basically non existent, since we only have an example with beacon packet.

pirib, 2019-10-07:
Thanks for heads-up, will do

closed 2019-10-07 by pirib