AdaCore/Ada_Drivers_Library #313: Support for STM32401RE board

BottCode, 2019-10-06:
Hi,

will STM32401RE be supported? It seems there's no driver for this board. 
How can I contribute in order to support this board?

Fabien-Chouteau, 2019-10-07:
Hi @BottCode ,

The `STM32F401` is close to the `STM32F407` so you should be able to use the same drivers.

BottCode, 2019-10-07:
Thanks for your answer @Fabien-Chouteau, I'll try to use `STM32F407`'s drivers as soon as possible.

FredPraca, 2020-09-19:
As the Nucleo 401RE seems to be really close to the Nucleo 411RE, I started to port the 411 I've done to the 401.
As soon as I'll get something, I'll need a beta tester so stay tuned

FredPraca, 2020-09-23:
Here we go !
I finished the port yesterday evening and it's available on my [nucleo_f401re branch](https://github.com/FredPraca/Ada_Drivers_Library/tree/nucleo_f401re).