AdaCore/Ada_Drivers_Library #325: U(S)ART support for STM32F7xx

ezwiebel, 2020-01-23:
Hello,

Is there a reason why the U(S)ART ports have been deactivated and no driver developed for them for the STM32F7xx platforms ?
I'm currently trying to develop one based on the F4 implementation and have troubles with RX link. Can't manage to read anything on the pin even though I'm sure something's incoming. I'm able to correctly write anything though...

Any ideas? Any help or reason would be much appreciated.

Cheers.

Fabien-Chouteau, 2020-01-24:
Hi @ezwiebel ,

The SMT32F7 USART are different than the STM32F4 ones. So the drivers are not compatible, and nobody developed the STM32F7 USART drivers yet.

I don't know this peripheral enough to help you here. Maybe you can compare the documentation of the F4 and the F7 to spot the differences.

ezwiebel, 2020-01-24:
Ok, thanks for the reply.
Was just wondering if someone tried and failed for some reason as USART/Serial is quite a common peripheral to use early in development (for debug purpose for example).
I already checked the documentation and SVDs and there are quite sensible differences in the register addressing and layout but nothing really out of reach apparently... but I'm still failing. 
Currently reading through C drivers to spot the miss !

I'll let you know how it goes if I success in implementing it => contribute for others ;)

emzaet, 2020-12-28:
hello I'm a begginer and I was learning ada. I want to learn about hardware with boards etc. But I don't know what cheap compatible board and processor with ada buy. And Is it possible to program TTL, IC chips,  with ada?

Fabien-Chouteau, 2021-01-04:
Hello @emzaet.

We have a list of examples for each of the boards supported by this project: https://github.com/AdaCore/Ada_Drivers_Library/tree/master/examples

closed 2021-01-04 by Fabien-Chouteau