AdaCore/Ada_Drivers_Library #359: Enable simple text IO in HiFive1_rev_B board?

rod-chapman, 2020-08-19:
Could we change the default runtime in HiFive1_rev_B from "zfp-rv32imac" to "zap-hifive1"? The latter implements System.Text_IO using UART0, and thus support GNAT.IO and Ada.Text_IO properly.

I think this just requires a change to scripts/config/boards.py to change the value of Runtime_Name_Suffix and then a rebuild of al the other auto-generated files.

Can anyone from AdaCore confirm that this is correct?

Fabien-Chouteau, 2020-09-01:
Hi @rod-chapman,

The problem is that the HiFive1 rev B doesn't have as much ROM as the rev A, so using the the zfp-hifive1 run-time can lead to more problems.

You can use [fe310-uart.ads](https://github.com/AdaCore/Ada_Drivers_Library/blob/master/arch/RISC-V/SiFive/drivers/fe310-uart.ads) to send text on the UART.

closed 2020-09-01 by Fabien-Chouteau

rod-chapman, 2020-09-01:
On 01/09/2020 13:38, Fabien Chouteau wrote:
>
> Closed #359 <https://github.com/AdaCore/Ada_Drivers_Library/issues/359>.
>
Indeed - I have a port of GNAT.IO working that uses FE310.UART and it 
seems to work well...

  Thanks,

   Rod

pat-rogers, 2020-09-08:
On 01-Sep-20 8:06 AM, Roderick Chapman wrote:
> On 01/09/2020 13:38, Fabien Chouteau wrote:
>  >
>  > Closed #359 <https://github.com/AdaCore/Ada_Drivers_Library/issues/359>.
>  >
> Indeed - I have a port of GNAT.IO working that uses FE310.UART and it
> seems to work well...

Can you share these, and all the other setup changes you've made?

rod-chapman, 2020-09-08:
On 08/09/2020 15:23, Pat Rogers wrote:
>
> Can you share these, and all the other setup changes you've made?

Sure... I am also working on getting access to the RV32 "cycle" and 
"instret" performance counters to get more accurate timing data.  Do I 
have permission(s) to open a Pull Request on the Ada_Drivers_Library repo?

  - Rod

Fabien-Chouteau, 2020-09-08:
Hi @rod-chapman, anyone has permission to open Pull Request form a fork of the repo.

rod-chapman, 2020-09-08:
OK... see new Issue #362 for the performance counters topic.