AdaCore/Ada_Drivers_Library #365: Access to FE310-specific performance counters

rod-chapman, 2020-09-16:
Following PR #362, we can now implement support for the micro-architectural performance counters implemented by the FE310 in the HiFive1. Only 2 such counters are implemented (mhpmcounter3 and 4), controlled by mhpmevent3 and 4 control registers, with specific meaning defined bu the FE310 Reference Manual.

rod-chapman, 2020-09-16:
Actually, the counters are specific to the FE31 RISC-V core, which is in the FE310-G000, FE310-G002 and FE310-G003 variants, so is common to all, so these can apply to both the HiFive1 and the HiFive1_RevB boards.

So... suggest that the counters are defined in a new public child package of FE310 - perhaps FE310.Perf ???

Fabien-Chouteau, 2020-09-16:
Hi Rod,

I prefer `FE310.Performance_Monitor` based on the documentation chapter name.

rod-chapman, 2020-09-17:
FE310.Performance_Monitor is good for me! Do you feel you can adapt the package spec that I sent you, or would you prefer to start from scratch?

Fabien-Chouteau, 2020-09-17:
> Do you feel you can adapt the package spec that I sent you, or would you prefer to start from scratch?

I thought you were going to do it :)

rod-chapman, 2020-09-18:
OK... I will have a go today. Is there a documented coding standard and dev process that I'm supposed to follow?

rod-chapman, 2020-09-18:
Hmmm...
```
RodPro2:drivers rchapman$ git push --set-upstream origin fe310_performance
remote: Permission to AdaCore/Ada_Drivers_Library.git denied to rod-chapman.
fatal: unable to access 'https://github.com/AdaCore/Ada_Drivers_Library.git/': The requested URL returned error: 403
```

Fabien - do you need to grant me permission to push?

rod-chapman, 2020-09-18:
I also sent the CLA (a PDF scan) to contributions@adacore.com

rod-chapman, 2020-09-18:
OK.. PR 370 is open. A first draft is there. I am trying to test it now...

PR #370 (2020-10-12): Add first draft of FE310.Performance_Monitor package.
Opening Pull Request.

Closes #365
closed 2020-10-07 by Fabien-Chouteau