AdaCore/Ada_Drivers_Library #374: install_dependencies.py appears to be broken

WickedShell, 2020-10-23:
I know I've used this script in the path with some slight issues and success, but it's no longer working.

``` zsh
wickedshell@milton ~/code % mkdir foo
wickedshell@milton ~/code % cd foo 
wickedshell@milton ~/code/foo % git clone https://github.com/AdaCore/Ada_Drivers_Library.git Ada_Drivers_Library/repo/
Cloning into 'Ada_Drivers_Library/repo'...
remote: Enumerating objects: 34, done.
remote: Counting objects: 100% (34/34), done.
remote: Compressing objects: 100% (31/31), done.
remote: Total 21139 (delta 10), reused 14 (delta 3), pack-reused 21105
Receiving objects: 100% (21139/21139), 18.58 MiB | 18.05 MiB/s, done.
Resolving deltas: 100% (13883/13883), done.
wickedshell@milton ~/code/foo % ./Ada_Drivers_Library/repo/scripts/install_dependencies.py                    
ROOT_DIR :/home/wickedshell/code/foo/Ada_Drivers_Library/repo
$ git clone https://github.com/AdaCore/bb-runtimes /home/wickedshell/code/foo/Ada_Drivers_Library/repo/bb-runtimes

Running build command:
$ /usr/bin/python3 /home/wickedshell/code/foo/Ada_Drivers_Library/repo/bb-runtimes/install.py --arch=arm-eabi --prefix=/home/wickedshell/opt/GNAT/2019-arm-elf/bin/../arm-eabi/lib/gnat
install runtime sources for stm32f4

Dependency install command error (returned 1):
Traceback (most recent call last):
  File "./build_rts.py", line 274, in <module>
    main()
  File "./build_rts.py", line 250, in main
    projects += installer.install(
  File "/home/wickedshell/code/foo/Ada_Drivers_Library/repo/bb-runtimes/support/bsp_sources/installer.py", line 159, in install
    runtime_sources = self._find_rts_sources(destination, rts_descriptor)
  File "/home/wickedshell/code/foo/Ada_Drivers_Library/repo/bb-runtimes/support/bsp_sources/installer.py", line 136, in _find_rts_sources
    assert ret is not None, "Cannot find %s" % rts_json_file
AssertionError: Cannot find rts-sources.json
Traceback (most recent call last):
  File "/home/wickedshell/code/foo/Ada_Drivers_Library/repo/bb-runtimes/install.py", line 71, in <module>
    main()
  File "/home/wickedshell/code/foo/Ada_Drivers_Library/repo/bb-runtimes/install.py", line 67, in main
    subprocess.check_call(cmd)
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python3', './build_rts.py', '--build', '--force', '--output=/home/wickedshell/opt/GNAT/2019-arm-elf/arm-eabi/lib/gnat', 'stm32f4', 'stm32f429disco', 'stm32f469disco', 'stm32f746disco', 'stm32756geval', 'stm32f769disco', 'samg55', 'sam4s', 'samv71', 'openmv2', 'rpi2', 'feather_stm32f405', 'cortex-m0', 'cortex-m0p', 'cortex-m1', 'cortex-m3', 'cortex-m4', 'cortex-m4f', 'cortex-m7f', 'cortex-m7df']' returned non-zero exit status 1.
```

I'm not sure what's supposed to be in `rts-sources.json` here at this point. I'm also unsure if this is more a problem with the bb-runtimes repo, or not having prepped all the files before trying to install the bb-runtimes.

Fabien-Chouteau, 2020-10-26:
This looks similar to: https://github.com/AdaCore/bb-runtimes/issues/42

See my answers there and make sure you have the compilers in your `PATH`.

closed 2020-10-26 by Fabien-Chouteau

BrentSeidel, 2020-10-26:
If you have multiple Ada installations if your PATH, they need to be in the correct order.  It appears that gprls only looks in the first Ada installation it finds.  This is on both MacOS and Windows 10.