AdaCore/Ada_Drivers_Library #375: Some examples do not build

astro-jerry, 2020-11-07:
Many of the examples do not build. In particular, the feather, HiFive1, HiFive1_rev_B, OpenMV2 and Unleashed.
The same is true in trying to build the runtimes in the bb-runtimes-2020-20200814-19A1B-src archive.

I am using Windows 10.

It is disappointing that these do not work.

Fabien-Chouteau, 2020-11-09:
Hello @astro-jerry,

Can you copy the error messages?
Also, did you build and install the extra run-times as explained in the examples README: https://github.com/AdaCore/Ada_Drivers_Library/tree/master/examples#build-and-install-the-run-times ?

astro-jerry, 2020-11-19:
To build the missing runtimes in Windows, the readme says run 'python ./scripts/install_dependencies.py' from shell - which shell?

from cmd shell:

C:\GNAT\Ada_Drivers_Library-master>python ./scripts/install_dependencies.py
Traceback (most recent call las
  File "./scripts/install_dependencies.py", line 77, in <module>
    os.path.join(os.path.dirname(shutil.which("arm-eabi-gnatls")),
AttributeError: 'module' object has no attribute 'which'

from Ubuntu on Windows in Windows Terminal:
jerry@OZ10:/mnt/c/GNAT/Ada_Drivers_Library-master$ python3 ./scripts/install_dependencies.py
Traceback (most recent call last):
  File "./scripts/install_dependencies.py", line 77, in <module>
    os.path.join(os.path.dirname(shutil.which("arm-eabi-gnatls")),
  File "/usr/lib/python3.8/posixpath.py", line 152, in dirname
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType

Here is what I get running gprls

PS C:\gnat\Ada_Drivers_Library-master> gprls -v --target=arm-eabi
gprconfig: can't find a toolchain for the following configuration:
gprconfig: language 'ada', target 'arm-eabi', default runtime

GPRLS Community 2020 (20200818) (x86_64-pc-mingw32)
Copyright (C) 2015-2020, AdaCore

Source Search Path:
   C:\gnat\Ada_Drivers_Library-master\\

Object Search Path:
   C:\gnat\Ada_Drivers_Library-master\\

Project Search Path:
   <Current_Directory>
   C:\GNAT\2020-arm-elf\arm-eabi\lib\gnat
   C:\GNAT\2020-arm-elf\arm-eabi\share\gpr
   C:\GNAT\2020-arm-elf\share\gpr
   C:\GNAT\2020-arm-elf\lib\gnat

astro-jerry, 2020-11-22:
Here is a screenshot of the errors when trying to do GPS build command on HiFive1 example:
[image]

Fabien-Chouteau, 2020-11-23:
Hi @astro-jerry ,

Look like the compiler is not installed correctly. You should have a `riscv32-elf-gcc` in `PATH`.

astro-jerry, 2020-11-23:
This is what I mean by the installation being confusing. I am on Windows 10. I installed the Windows native compiler and then the ARM-elf compiler. I then put those two bin directories in my path. Trying to build the HiFive1 example from the Ada_drivers_Library, I get this problem. Is the HiFive not supported under Windows? Also I recently got the Windows downloads from my Academic account and there was a version from 8-18-2020 but now it is gone - was there something wrong with that version?

Then if I try to build and install the missing runtimes as described in the examples Readme, I get this error:
from powershell:
PS C:\Gnat\Ada_Drivers_Library-master> python ./scripts/install_dependencies.py
Traceback (most recent call last):
  File "./scripts/install_dependencies.py", line 77, in <module>
    os.path.join(os.path.dirname(shutil.which("arm-eabi-gnatls")),
AttributeError: 'module' object has no attribute 'which'

and trying to add the hfive support as described in the bb-runtimes-community-2020 Readme, I get these errors:

PS C:\Gnat\bb-runtimes-community-2020> build_rts.py --output=temp --build hifive1
install runtime sources for hifive1
Traceback (most recent call last):
  File "C:\GNAT\bb-runtimes-community-2020\build_rts.py", line 274, in <module>
    main()
  File "C:\GNAT\bb-runtimes-community-2020\build_rts.py", line 250, in main
    projects += installer.install(
  File "C:\GNAT\bb-runtimes-community-2020\support\bsp_sources\installer.py", line 159, in install
    runtime_sources = self._find_rts_sources(destination, rts_descriptor)
  File "C:\GNAT\bb-runtimes-community-2020\support\bsp_sources\installer.py", line 136, in _find_rts_sources
    assert ret is not None, "Cannot find %s" % rts_json_file
AssertionError: Cannot find rts-sources.json
PS C:\Gnat\bb-runtimes-community-2020>

I have mostly been using the 2017 versions of GNAT for ARM until now because I have built a lot of runtimes for various ST boards using the approach used in the 2016-2017 versions and didn't want to change everything each year with your new methods of handling the runtimes but I wanted some of your newer ones so I installed your latest versions on a new computer and then ran into all these problems! I even have a H743 Nucleo board runtime working some at 240MHz using the old versions but I am getting nowhere with this new version.

Fabien-Chouteau, 2020-11-30:
This HiFive one is a RISC-V32 board, so you need a riscv32-elf toolchain. Unfortunately we only provide them for Linux in the community release.

closed 2020-11-30 by Fabien-Chouteau