AdaCore/Ada_Drivers_Library #381: Target not found for Micro:Bit with Debian Linux

MarcusE1W, 2021-02-19:
Hi.

I struggle to build the first example of the [Micro:Bit Tutorial](https://blog.adacore.com/ada-for-microbit-part-1-getting-started).

I am using Armbian, a Debian variant on a Pinebook Pro with ARM processor. Installed is gnat-9, gprbuild and gnat-gps. That works fine for native Ada programms.

To use the Micro:Bit example I have added:
```
sudo apt install gnat-9-arm-linux-gnueabi
sudo apt install gcc-arm-none-eabi
```
This is guesswork on my side, not sure if this is right

This gives me this result with gprconfig:
```
gprconfig --show-targets
List of targets supported by a compiler:
aarch64-linux-gnu
arm-none-eabi
llvm
aarch64-linux-gnu (native target)
arm-linux-gnueabi
```

Unfortunately, if I try to build the `text_scrolling` example I get this error message:
```
gprbuild --target=arm-eabi -d -P/home/mmw/adl/examples/MicroBit/text_scrolling/text_scrolling.gpr -XADL_BUILD_CHECKS=Disabled -XADL_BUILD=Debug
gprconfig: can't find a toolchain for the following configuration:
gprconfig: language 'ada', target 'arm-eabi', runtime 'zfp-cortex-m0'
microbit_zfp.gpr:61:25: warning: libraries are not supported on this platform
microbit_zfp.gpr:2:17: no compiler for language "Asm_CPP", cannot compile "crt0.S"
gprbuild: *** compilation phase failed
```

Any ideas how to progress welcome.

Fabien-Chouteau, 2021-02-19:
Hello @MarcusE1W ,

As stated in the blog post, you first have to download and install GNAT Community from [adacore.com/download](https://adacore.com/download).
You will need both the x86_64 and arm-elf packages.

Regards,

closed 2021-02-19 by Fabien-Chouteau

MarcusE1W, 2021-02-19:
Thanks. So you mean the example only works for the GNAT Community edition ? Is that the case for the whole ADL ?
As mentioned I use a Pinebook with ARM processor as host and if I don't get it wrong, there is no GNAT Community support for linux on aarch64 architecture, right ?

Fabien-Chouteau, 2021-02-19:
In theory it ADL works with any GNAT arm-eabi, you can either use GNAT Community or build your own cross GNAT from the GCC sources.