AdaCore/Ada_Drivers_Library #383: Project Wizard doesn't generate startup files

JCGobbi, 2021-03-21:
When running `project_wizard.py` and choosing to use `startup-gen` to generate the `link.ld` and `crt0.S` files, the script uses the file `ada_drivers_library.gpr` instead of the "real" project file, that must be outside the `Ada_Drivers_Library` directory.

Considering that the `Ada_Drivers_Library` directory is inside the actual project directory, the text for `startup-gen` inside `project_wizard.py` could be:

`# Run startup-gen`
`    if config.get_config ("Use_Startup_Gen") == "True":`

`# Start changes`
`        prj_dir = os.path.abspath(os.path.join(project_directory, ".."))`
`        print("\nPut the full name of the ZFP project file.")`
`        config.query_string_key('ZFP_Project_File', default="zfp.gpr")`
`        prj_path = os.path.join(prj_dir, config.get_config("ZFP_Project_File"))`
`        print(" -> Generating startup files.")`
`        run_program ('startup-gen',`
`                     '-P', prj_path,`
`# End changes`

`                     '-l', os.path.join(source_dir, "link.ld"),`
`                     '-s', os.path.join(source_dir, "crt0.S"))`

This time, considering that `startup-gen` is in the `PATH`, it will use the `../zfp.gpr` project file or the name the user choose. I have tested and it runs fine. Please consider this like a suggestion.

Fabien-Chouteau, 2021-03-23:
Hello @JCGobbi,

I am not sure to understand the problem here. I think it is normal that startup-gen is called on the `ada_drivers_library.gpr` project.

JCGobbi, 2021-03-23:
Hello @Fabien-Chouteau,

I will explain better.

I have followed the instructions for compiling and using `startup-gen` and already generated startup files for the STM32F334R8 processor. This program needs a `*.gpr` file with the `package Device_Configuration` with memory descriptions, like kind, address and size. So we need to create a project file with that information, and it works fine.

Now, to generate an `ada_drivers_library.gpr` file with `project_wizard.py` with the option to use `startup-gen`, there must be in some place that same `package Device_Configuration` information.

If `startup-gen` uses the `ada_drivers_library.gpr` file to generate the startup files, then it will not generate them because this file is not there for the first time we run `project_wizard.py`. And if we create an `ada_drivers_library.gpr` file with the `package Device_Configuration` information, the wizard overwrites it.

So, my suggestion is to use an external `*.gpr` file with `package Device_Configuration` information for `startup-gen` run from inside `project_wizard.py`.

Fabien-Chouteau, 2021-03-24:
No problem :+1:

closed 2021-03-24 by Fabien-Chouteau