AdaCore/Ada_Drivers_Library #397: EXAMPLES: Installing run-time dependencies fails with both Python 2 and Python 3 on Linux

gregkrsak, 2022-02-10:
(Operating System: Lubuntu 21.10, with full uname, python, and git version outputs noted below)

How to reproduce:
1. Follow the instructions on line 69 of the examples readme: https://github.com/AdaCore/Ada_Drivers_Library/blame/master/examples/README.md#L69
2. Observe the output below, for either Python 2 or Python 3.

What am I not thinking of?

(base) gregk@imaginebook:~/Code/Ada_Drivers_Library$ ```python ./scripts/install_dependencies.py```
```
Traceback (most recent call last):
  File "/home/gregk/Code/Ada_Drivers_Library/./scripts/install_dependencies.py", line 77, in <module>
    os.path.join(os.path.dirname(shutil.which("arm-eabi-gnatls")),
  File "/opt/anaconda3/lib/python3.9/posixpath.py", line 152, in dirname
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType
```

(base) gregk@imaginebook:~/Code/Ada_Drivers_Library$ ```python2 ./scripts/install_dependencies.py```
```
Traceback (most recent call last):
  File "./scripts/install_dependencies.py", line 77, in <module>
    os.path.join(os.path.dirname(shutil.which("arm-eabi-gnatls")),
AttributeError: 'module' object has no attribute 'which'
```

(base) gregk@imaginebook:~/Code/Ada_Drivers_Library$ ```python3 ./scripts/install_dependencies.py```                                                                                                                                                                 
```
Traceback (most recent call last):
  File "/home/gregk/Code/Ada_Drivers_Library/./scripts/install_dependencies.py", line 77, in <module>
    os.path.join(os.path.dirname(shutil.which("arm-eabi-gnatls")),
  File "/opt/anaconda3/lib/python3.9/posixpath.py", line 152, in dirname 
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType
```

(base) gregk@imaginebook:~/Code/Ada_Drivers_Library$ ```python2```
```
Python 2.7.18 (default, Sep 10 2021, 14:59:31) 
[GCC 11.2.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> quit()
```

(base) gregk@imaginebook:~/Code/Ada_Drivers_Library$ ```python3```
```
Python 3.9.7 (default, Sep 16 2021, 13:09:58) 
[GCC 7.5.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> quit()
```

(base) gregk@imaginebook:~/Code/Ada_Drivers_Library$ ```uname -a```
```
Linux imaginebook 5.13.0-28-generic #31-Ubuntu SMP Thu Jan 13 17:41:06 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

(base) gregk@imaginebook:~/Code/Ada_Drivers_Library$ ```git --version```
```
git version 2.32.0
```

Repository was cloned using: ```git clone --recursive https://github.com/AdaCore/Ada_Drivers_Library.git```                                                                                                                                                              
```
Cloning into 'Ada_Drivers_Library'...
remote: Enumerating objects: 21404, done.
remote: Counting objects: 100% (239/239), done.
remote: Compressing objects: 100% (153/153), done.
remote: Total 21404 (delta 95), reused 155 (delta 58), pack-reused 21165
Receiving objects: 100% (21404/21404), 18.68 MiB | 5.51 MiB/s, done. 
Resolving deltas: 100% (13998/13998), done.
```

Fabien-Chouteau, 2022-02-14:
Hi @gregkrsak,

I just tried with both Python 3.8 and 3.9 on my Ubuntu 20.04 and it works.

`/opt/anaconda3/lib/python3.9/posixpath.py` doesn't look like a standard path for python libs. I don't know what `anaconda3` is, but it could be the source of the issue here.

gregkrsak, 2022-02-14:
Good point. I'll look into this, and re-open the issue if it persists. Thanks for the quick response.

closed 2022-02-14 by gregkrsak