AdaCore/Ada_Drivers_Library #403: Many warnings with GCC 12

simonjwright, 2022-04-12:
There are quite a few new warnings when compiling the current library with GCC 12. I'm only set up for STM32(F4) and Microbit, and below are the issues I see.

I understand that the `pragma Unreferenced` issue (now intended for variables that are deliberately not referenced at all, e.g. an RAII lock) is best addressed by changing the variable's name from e.g. `Status` to `Unused`, or `Dummy`, or `Status_Unused`, or `Unused_Status` ...

A (renamed) variable that isn't referenced should just be deleted.

I would offer to make these changes, but you might prefer to hold off until the problem can be fixed for all the supported devices?

```
ak8963.adb:144:10: warning: pragma Unreferenced given for "Status" [enabled by default]
cs43l22.adb:50:27: warning: pragma Unreferenced given for "Status" [enabled by default]
ft5336.adb:349:52: warning: pragma Unreferenced given for "Status" [enabled by default]
ft6x06.adb:80:52: warning: pragma Unreferenced given for "Status" [enabled by default]
nrf-gpio-tasks_and_events.adb:102:07: warning: renamed variable "INTENSET" is not referenced [-gnatwu]
nrf-gpio-tasks_and_events.adb:116:07: warning: renamed variable "CONFIG" is not referenced [-gnatwu]
nrf-gpio-tasks_and_events.adb:132:07: warning: renamed variable "INTENSET" is not referenced [-gnatwu]
nrf-gpio-tasks_and_events.adb:147:07: warning: renamed variable "CONFIG" is not referenced [-gnatwu]
nrf-gpio-tasks_and_events.adb:80:07: warning: renamed variable "INTENCLR" is not referenced [-gnatwu]
nrf-gpio-tasks_and_events.adb:91:07: warning: renamed variable "INTENCLR" is not referenced [-gnatwu]
nrf-gpio.adb:154:07: warning: renamed variable "CNF" is not referenced [-gnatwu]
nrf-gpio.adb:54:07: warning: renamed variable "CNF" is not referenced [-gnatwu]
stmpe1600.adb:212:46: warning: pragma Unreferenced given for "Status" [enabled by default]
stmpe1600.adb:241:46: warning: pragma Unreferenced given for "Status" [enabled by default]
stmpe1600.adb:259:45: warning: pragma Unreferenced given for "Status" [enabled by default]
stmpe1600.adb:266:46: warning: pragma Unreferenced given for "Status" [enabled by default]
stmpe1600.adb:283:45: warning: pragma Unreferenced given for "Status" [enabled by default]
stmpe1600.adb:301:46: warning: pragma Unreferenced given for "Status" [enabled by default]
stmpe1600.adb:308:47: warning: pragma Unreferenced given for "Status" [enabled by default]
vl53l0x.adb:1406:19: warning: pragma Unreferenced given for "Status" [enabled by default]
vl53l0x.adb:880:53: warning: pragma Unreferenced given for "Status" [enabled by default]
vl53l0x.adb:894:54: warning: pragma Unreferenced given for "Status" [enabled by default]
vl53l0x.adb:895:64: warning: pragma Unreferenced given for "Status" [enabled by default]
vl53l0x.adb:980:67: warning: pragma Unreferenced given for "Status" [enabled by default]
wm8994.adb:62:27: warning: pragma Unreferenced given for "Status" [enabled by default]
```

Fabien-Chouteau, 2022-04-12:
Hi @simonjwright, GCC 12 is not quite on my radar yet and I am not sure when/how we will port Ada Drivers Library.
If you have changes that are compatible with both GCC 12 and CE 2021 don't hesitate to send patches :+1:

simonjwright, 2022-04-12:
I can't do the CE 2021 part (not without building the arm-eabi cross) so that won't be happening.

closed 2022-04-12 by simonjwright