AdaCore/Ada_Drivers_Library #411: RISC-V requires "zicsr" ISA extension to build with FSF GNAT 12.1.1

rod-chapman, 2022-06-29:
A recent change in GNU binutils (see https://groups.google.com/a/groups.riscv.org/g/sw-dev/c/aE1ZeHHCYf4) means that the RISC-V "performance counter" registers are now in a specific ISA extension called "zicsr". This change is shipping with the FSF GNAT 12.1.1 in Alire, so the build of Ada_Drivers_Library for the HiFive_revb (for example) fails. The project files needs to be updated to select "-march=rv32imac_zicsr" for all units, or (more generally) to add "_zicsr" to the end of the "-march" option for units that require the CSRR instructions.

rod-chapman, 2022-06-30:
It may also be possible to directly insert an assembler directive in the generated code for the packages that require the CSRR and CSRW instructions. I think you need to add
 .option arch,+zicsr

near the top of the generated .s file.  Can this be done with package Machine_Code?

Details in the RISC-V specific section of the most recent GNU as manual.

rod-chapman, 2022-06-30:
For reference, my local changes to SPARKNaCl to get GNAT 12.1.1 working are here:
 https://github.com/rod-chapman/SPARKNaCl/pull/28

Fabien-Chouteau, 2022-06-30:
`.option arch,+zicsr` in inlined machine code would be the best option in my opinion. Did you try it?

Fabien-Chouteau, 2022-07-01:
Confirmed, I started build for a new release of the toolchains with binutils 2.38.

rod-chapman, 2022-07-04:
OK... let me know when it's ready (and how to get it) and I will give it a try...

rod-chapman, 2022-07-22:
The SPARKNaCl performance test program now seems to build OK with the new GNAT 12.1.2 with no modification to the assembly language component(s), so looking good.

Fabien-Chouteau, 2022-07-22:
Without the `.option arch,+zicsr`?

Thanks for the update,

closed 2022-07-22 by Fabien-Chouteau

rod-chapman, 2022-07-22:
Yes... my sources (see SPARKNaCl/perf/csrs.s) are unmodified and it builds OK...