AdaCore/Ada_Drivers_Library #420: Extension zicsr required in HiFive1_rev_B board

rocher, 2023-01-31:
Hi,

I'm trying to compile the example provided for the HiFive1_rev_B board, but I'm getting the same errors mentioned in [https://github.com/AdaCore/Ada_Drivers_Library/issues/411](url):

> Ada_Drivers_Library/arch/RISC-V/src/riscv-csr_generic.adb:55: Error: unrecognized opcode `csrw mhpmevent3,a5', extension `zicsr' required

I'm using `alr 1.2.2` and `gnat_riscv64_elf 12.2.1`, with the proper target and runtime for this board.
Any idea or suggestion?

rod-chapman, 2025-01-16:
I have the same problem trying to reproduce the performance results for SPARKNaCl.

I'm using gnat_riscv64_elf_13.2.1

I get
```
gprbuild -Ptsign -v -XSPARKNACL_RUNTIME_MODE=zfp -XSPARKNACL_RUNTIME_CHECKS=disabled -XSPARKNACL_CONTRACTS=disabled  -XSPARKNACL_TARGET_ARCH=rv32imc_a4 -XSPARKNACL_BUILD_MODE=debug
Changing to object directory of "HiFive1_rev_B_ZFP": "/home/rchapman/prog/AdaCore/Ada_Drivers_Library/boards/HiFive1_rev_B/obj/zfp_Debug/"
/home/rchapman/.config/alire/cache/dependencies/gnat_riscv64_elf_13.2.1_938f208c/bin/riscv64-elf-gcc -c -x ada -gnatA --RTS=/home/rchapman/.config/alire/cache/dependencies/gnat_riscv64_elf_13.2.1_938f208c/riscv64-elf/lib/gnat/light-rv32imac/ -march=rv32imac -mabi=ilp32 -fno-tree-loop-distribute-patterns -O0 -gnata -fcallgraph-info=su -fcallgraph-info=su -g -gnatwa -gnatw_A -gnatQ -gnatw.X -ffunction-sections -fdata-sections -gnatec=/tmp/GNAT-TEMP-000002.TMP -gnatem=/tmp/GNAT-TEMP-000003.TMP /home/rchapman/prog/AdaCore/Ada_Drivers_Library/arch/RISC-V/SiFive/drivers/fe310-performance_monitor.adb
/home/rchapman/prog/AdaCore/Ada_Drivers_Library/arch/RISC-V/src/riscv-csr_generic.adb: Assembler messages:
/home/rchapman/prog/AdaCore/Ada_Drivers_Library/arch/RISC-V/src/riscv-csr_generic.adb:55: Error: unrecognized opcode `csrw mhpmevent3,a5', extension `zicsr' required
```

Note the -march=rv32imac in that compiler invocation. I know I need to change this to -march=rv32imac_zcisr but where is this setting stored?

I see those options in the compiler installation in files called "runtime.xml" - but which one needs to change and how?

rod-chapman, 2025-01-16:
I think I have it working now - I needed to modify the runtime.xml file in the light_rv32imac directory in the compiler distro. Does that sound right?

Fabien-Chouteau, 2025-01-16:
Hi @rod-chapman, That's what I was about to say. The architecture switches are set in the run-time.

There might be two other options:

1. We could change the script that generates the ADL project file to include `-march=rv32imac_zicsr `
2. We could add the `.option arch, +zicsr` in the CSR inline assembly code.

rod-chapman, 2025-01-16:
I'm not sure if setting that arch switch should be part of the standard runtime distro of the compiler, or part of the config of the runtime library for that particular board? Is the "light_rv32imac" runtime used for any CPUs/boards that _don't_ have the zicsr extension?

Fabien-Chouteau, 2025-01-16:
> Is the "light_rv32imac" runtime used for any CPUs/boards that _don't_ have the zicsr extension?

We are supposed to have one "light" run-time for every architecture, but for RISC-V, there are so many combinations that it is practically impossible:
 
- light_rv32i
- light_rv32i_zicsr
- light_rv32i_zifencei
- light_rv32i_zicsr_zifencei
- light_rv32imc
- light_rv32imc_zicsr
- light_rv32imc_zifencei
- light_rv32imc_zicsr_zifencei
- light_rv32imac
- light_rv32imac_zicsr
- light_rv32imac_zifencei
- light_rv32imac_zicsr_zifencei
- etc...

One of the reasons why I made this little experiment here to use with Alire: https://alire.ada.dev/crates/bare_runtime