AdaCore/Ada_Drivers_Library #422: How to get the toolchain up and running?

louis-etne, 2023-04-05:
Hi,
I'm trying to compile any example from this project, but it seems that I can't get the toolchain correctly installed:

```
Ada_Drivers_Library/examples/STM32F429_Discovery
$ gprbuild -P blinky_f429disco.gpr 
gprconfig: can't find a toolchain for the following configuration:
gprconfig: language 'ada', target 'arm-eabi', runtime 'embedded-stm32f429disco'
stm32f429_discovery_full.gpr:61:25: warning: libraries are not supported on this platform
common.gpr:1:09: no compiler for language "Ada", cannot compile "last_chance_handler.adb"
gprbuild: *** compilation phase failed
```
I downloaded the [gnat-arm](https://github.com/alire-project/GNAT-FSF-builds/releases), as required by the [README](https://github.com/AdaCore/Ada_Drivers_Library/blob/master/examples/README.md):

```
$ arm-eabi-gnat --version
GNAT 12.2.0
Copyright (C) 1996-2022, Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
What did I missed? I have the same error in gnatstudio.

pat-rogers, 2023-04-10:
Hi,

What does the output of the following command report:

    arm-eabi-gnatls -v --RTS=embedded-stm32f4

Best regards,

Pat Rogers

--
Patrick Rogers, PhD
***@***.***

louis-etne, 2023-04-11:
Hi,
Thanks for your answer!

This is what i get:
```
$ arm-eabi-gnatls -v --RTS=embedded-stm32f4

GNATLS 12.2.0
Copyright (C) 1997-2022, Free Software Foundation, Inc.

Source Search Path:
   <Current_Directory>
   /home/louis/.local/bin/gnat-arm-elf-linux64-12.2.0-1/arm-eabi/lib/gnat/embedded-stm32f4/gnat_user
   /home/louis/.local/bin/gnat-arm-elf-linux64-12.2.0-1/arm-eabi/lib/gnat/embedded-stm32f4/gnarl_user
   /home/louis/.local/bin/gnat-arm-elf-linux64-12.2.0-1/arm-eabi/lib/gnat/embedded-stm32f4/gnat
   /home/louis/.local/bin/gnat-arm-elf-linux64-12.2.0-1/arm-eabi/lib/gnat/embedded-stm32f4/gnarl

Object Search Path:
   <Current_Directory>
   /home/louis/.local/bin/gnat-arm-elf-linux64-12.2.0-1/arm-eabi/lib/gnat/embedded-stm32f4/adalib

Project Search Path:
   <Current_Directory>
   /home/louis/.local/bin/gnat-arm-elf-linux64-12.2.0-1/arm-eabi/embedded-stm32f4/lib/gnat
   /home/louis/.local/bin/gnat-arm-elf-linux64-12.2.0-1/arm-eabi/embedded-stm32f4/share/gpr
   /home/louis/.local/bin/gnat-arm-elf-linux64-12.2.0-1/arm-eabi/lib/gnat
   /home/louis/.local/bin/gnat-arm-elf-linux64-12.2.0-1/arm-eabi/share/gpr
   /home/louis/.local/bin/gnat-arm-elf-linux64-12.2.0-1/share/gpr
   /home/louis/.local/bin/gnat-arm-elf-linux64-12.2.0-1/lib/gnat
```

pat-rogers, 2023-04-11:
On 11-Apr-23 10:57 AM, Louis Etienne wrote:
> Hi,
> Thanks for your answer!

You're very welcome!

> This is what i get:
> 
> |$ arm-eabi-gnatls -v --RTS=embedded-stm32f4 GNATLS 12.2.0 Copyright (C) 

OK, that's all good, so the toolchain installation itself seems good to 
go, although not necessarily all the runtimes you might want. But I 
still am not convinced that it is as expected.

If you try to build the blinky project for the STM32F4 Disco board (not 
the 429), does that work?  In the STM32F4_DISCO directory, try this:

    gprbuild -P blinky_f4disco.gpr

This will try to build with a different target board and runtime, as a test.

Best regards,

Pat Rogers

--
Patrick Rogers, PhD
***@***.***

louis-etne, 2023-04-12:
Hi,
I tried every example without any success and the same error each time:
```
language 'ada', target 'arm-eabi', runtime 'embedded-stm32f429disco'
```
(the target and the runtime changed of course, but the error was the same)

pat-rogers, 2023-04-12:
On 12-Apr-23 12:09 PM, Louis Etienne wrote:
> Hi,
> I tried every example without any success and the same error each time:
> 
> |language 'ada', target 'arm-eabi', runtime 'embedded-stm32f429disco' |
> 
> (the target and the runtime changed of course, but the error was the same)

OK, thanks for trying that one.

Let's get some more info. Add "-v" to the end of the gprbuild command 
(without the quotes). What does that display?

pat-rogers, 2023-04-12:
Hi Louis,

For the sake of comparison to your setup, I did the following to verify 
the downloads and overall approach.

I created a new Ubuntu virtual machine.

In that VM I created /usr/gnat/arm-elf/ (via the sudo command). The 
directory names themselves are not important, it's just what I am used to.

I downloaded gnat-arm-elf-linux64-12.2.0-1.tar.gz for the toolchain 
itself and expanded/untarred it into the above, resulting in these 
subdirectories:

├── arm-eabi

├── bin

├── include

├── lib

├── lib64

├── libexec

└── share

I then put /usr/gnat/arm-elf/bin/ on the PATH in my .profile file. 
Instead of just doing "source .profile" to update the PATH value I 
logged off and then back on, so that the effect would be global across 
the couple of terminal windows I wanted to use.

Next I downloaded gprbuild-x86_64-linux-22.0.0-1.tar.gz for gprbuild, 
expanded it locally, CD'd into it, and then ran the installer:

    sudo ./doinstall

That adds more files under the same subdir names as the compiler, ie 
bin/, libexec/, and share/. Therefore I took the installer option to put 
it in a specific location, and specified the same location as the 
compiler, in other words /usr/gnat/arm-elf/. As a result it added the 
gprbuild-specific files to those existing subdirs.

Next, I cloned the Ada Drivers Library. Once done, I opened a terminal 
at Ada_Drivers_Library/examples/STM32F4_DISCO and built the blinky example:

    gprbuild -P blinky_f4disco.gpr

That all worked as expected. I then did the same with the F429 Disco 
blinky example.

So, how does your installation differ from the above?

Best regards,

Pat Rogers

--
Patrick Rogers, PhD
***@***.***

louis-etne, 2023-04-14:
Hi,
with more information I get:
```
$ which gprbuild
/usr/bin/gprbuild
$ gprbuild -v -P blinky_f4disco.gpr 
gprconfig: can't find a toolchain for the following configuration:
gprconfig: language 'ada', target 'arm-eabi', runtime 'embedded-stm32f4'

==============Messages for file: /home/louis/Documents/projects/embedded/Ada_Drivers_Library/boards/stm32f407_discovery/stm32f407_discovery_full.gpr
    61.    for Library_Name use "ada_drivers_library";
                                |
        >>> warning: libraries are not supported on this platform

==============Messages for file: /home/louis/Documents/projects/embedded/Ada_Drivers_Library/examples/shared/common/common.gpr
     1. project Common is
                |
        >>> no compiler for language "Ada", cannot compile "last_chance_handler.adb"
gprbuild: *** compilation phase failed
```

I have exactly the same folder structure for my gnat-arm toolchain. I installed it in ~/.local/bin and then added it to my path in my .bashrc. 
I reinstalled gprbuild, not using dnf (I'm on fedora) but via the installer, to have the exact same version as you (and a more recent one). I still have the issue:

```
$ which gprbuild
/usr/local/bin/gprbuild
$  gprbuild -v -P blinky_f4disco.gpr
gprconfig: can't find a toolchain for the following configuration:
gprconfig: language 'ada', target 'arm-eabi', runtime 'embedded-stm32f4'

==============Messages for file: /home/louis/Documents/projects/embedded/Ada_Drivers_Library/boards/stm32f407_discovery/stm32f407_discovery_full.gpr
    61.    for Library_Name use "ada_drivers_library";
                                |
        >>> warning: libraries are not supported on this platform

==============Messages for file: /home/louis/Documents/projects/embedded/Ada_Drivers_Library/examples/shared/common/common.gpr
     1. project Common is
                |
        >>> no compiler for language "Ada", cannot compile "last_chance_handler.adb"
gprbuild: *** compilation phase failed
```

Again, thanks for your time.

pat-rogers, 2023-04-14:
Hi,

Thanks for trying that experiment. This is very strange indeed, but it 
is almost certainly a setup issue in some form. Something is conflicting.

Please issue the following command and send the resulting output (it 
will be a lot of info, hopefully):

    gprconfig --batch --target=arm-elf --config=ada,, -v

Best regards,

Pat Rogers

--
Patrick Rogers, PhD
***@***.***

louis-etne, 2023-04-14:
As requested:

```
$  gprconfig --batch --target=arm-elf --config=ada,, -v
Parsing knowledge base at /usr/local/share/gprconfig
Parsing file /usr/local/share/gprconfig/0_default.xml
Parsing file /usr/local/share/gprconfig/aarch64-vx7.xml
Parsing file /usr/local/share/gprconfig/aarch64-vx7r2.xml
Parsing file /usr/local/share/gprconfig/arm-vx6.xml
Parsing file /usr/local/share/gprconfig/arm-vx7.xml
Parsing file /usr/local/share/gprconfig/arm-vx7r2.xml
Parsing file /usr/local/share/gprconfig/asis.xml
Parsing file /usr/local/share/gprconfig/asm.xml
Parsing file /usr/local/share/gprconfig/baselined.xml
Parsing file /usr/local/share/gprconfig/bin_img.xml
Parsing file /usr/local/share/gprconfig/c.xml
Parsing file /usr/local/share/gprconfig/clean.xml
Parsing file /usr/local/share/gprconfig/compilers-arm-vx7.xml
languages: constant := C
Parsing file /usr/local/share/gprconfig/compilers-asis.xml
languages: constant := Ada
Parsing file /usr/local/share/gprconfig/compilers-baselined.xml
languages: constant := C++
languages: constant := C++
languages: constant := C
languages: constant := C
Parsing file /usr/local/share/gprconfig/compilers-e500v2-vx6.xml
warning: environment variable 'WIND_BASE' is not defined
languages: constant := C
warning: environment variable 'WIND_BASE' is not defined
languages: constant := C++
warning: environment variable 'WIND_BASE' is not defined
languages: constant := C
warning: environment variable 'WIND_BASE' is not defined
languages: constant := C++
Parsing file /usr/local/share/gprconfig/compilers-e500v2-vx653.xml
warning: environment variable 'WIND_BASE' is not defined
languages: constant := C++
warning: environment variable 'WIND_BASE' is not defined
languages: constant := C++
Parsing file /usr/local/share/gprconfig/compilers-e500v2-vx7.xml
languages: constant := C
Parsing file /usr/local/share/gprconfig/compilers-e500v2-vx7r2.xml
languages: constant := C
Parsing file /usr/local/share/gprconfig/compilers-ppc-vx653.xml
warning: environment variable 'WIND_BASE' is not defined
languages: constant := C
warning: environment variable 'WIND_BASE' is not defined
languages: constant := C++
Parsing file /usr/local/share/gprconfig/compilers-ppc-vx7.xml
languages: constant := C
Parsing file /usr/local/share/gprconfig/compilers-ppc-vx7r2.xml
languages: constant := C
Parsing file /usr/local/share/gprconfig/compilers-ppc64-vx7.xml
languages: constant := C
Parsing file /usr/local/share/gprconfig/compilers-ppc64-vx7r2.xml
languages: constant := C
Parsing file /usr/local/share/gprconfig/compilers-vxworks.xml
warning: environment variable 'WIND_BASE' is not defined
languages: constant := C
warning: environment variable 'WIND_BASE' is not defined
languages: constant := C++
warning: environment variable 'WIND_BASE' is not defined
languages: constant := C
warning: environment variable 'WIND_BASE' is not defined
languages: constant := C++
languages: constant := C
languages: constant := C++
Parsing file /usr/local/share/gprconfig/compilers-x86-vx7.xml
languages: constant := C
[... 195 lines cut ...]
      Filter=ada,,,,: language does not match
      Filter=ada,,,,: language does not match
    GCC_Pro is candidate: filename=arm-none-eabi-gcc
      target: executing "arm-none-eabi-gcc -dumpmachine" output=<use -v -v> no match
      target: grep matched="arm-none-eabi"
      version: executing "arm-none-eabi-gcc -v" output=<use -v -v> no match
      version: grep no match
      Ignore compiler, since couldn't guess its version
    GCC-ASM is candidate: filename=arm-none-eabi-gcc
      target: executing "arm-none-eabi-gcc -dumpmachine" output=<use -v -v> no match
      target: grep matched="arm-none-eabi"
      version: executing "arm-none-eabi-gcc -v" output=<use -v -v> no match
      version: grep matched="12.2.0"
      languages: constant := Asm,Asm_Cpp,Asm2
      Filter=ada,,,,: language does not match
      Filter=ada,,,,: language does not match
      Filter=ada,,,,: language does not match
      Filter=ada,,,,: language does not match
      Filter=ada,,,,: language does not match
      Filter=ada,,,,: language does not match
  Foreach compiler in /usr/sbin regexp=TRUE extra_dir=FALSE
gprconfig: can't find a toolchain for the following configuration:
gprconfig: language 'ada', target 'arm-elf', default runtime
Creating configuration file: arm-elf.cgpr
```

pat-rogers, 2023-04-15:
On 14-Apr-23 1:05 PM, Louis Etienne wrote:
> As requested:
> 
> |$ gprconfig --batch --target=arm-elf --config=ada,, -v Parsing 
> knowledge base at /usr/local/share/gprconfig Parsing file 

For some reason, on your installation gprconfig does not see all the 
runtimes included with the arm-elf compiler. The output I get in my VM 
from that same command is much, much more extensive, as a result.

In particular there should be many runtimes located here:

    <cross compiler installation root>/arm-eabi/lib/

In my case:

    /usr/gnat/arm-elf/arm-eabi/lib/

and gprconfig should find them, but it does not on your installation. 
Your output just has this:

> GNAT is candidate: filename=arm-eabi-gnatls
>       target: executing "arm-eabi-gcc -dumpmachine" output=<use -v -v> no match
>       target: grep matched="arm-eabi"
>       version: executing [...]

Do you have those runtime directories under arm-eabi/lib/ ?

Assuming so, there must be some other installation causing conflict.

At this point I am grasping at straws: I suggest renaming directories on 
the PATH, iteratively, to see if one of them contains the culprit.

pat-rogers, 2023-04-15:
On 14-Apr-23 1:05 PM, Louis Etienne wrote:
> As requested:
> 
> |$ gprconfig --batch --target=arm-elf --config=ada,, -v Parsing 
> knowledge base at /usr/local/share/gprconfig Parsing file 
> ...

> Creating configuration file: arm-elf.cgpr 

Be sure to delete any such .cgpr files you find in the dirs in which 
you've run the gprconfig tool.

pat-rogers, 2023-04-15:
On 15-Apr-23 2:16 PM, Pat Rogers wrote:
> On 14-Apr-23 1:05 PM, Louis Etienne wrote:
>
>
> Be sure to delete any such .cgpr files you find in the dirs in which
> you've run the gprconfig tool.

For that matter, is it possible that there is an existing .cgpr file 
that is in play? That would certainly explain things.

pat-rogers, 2023-04-16:
Just to be certain there was nothing I was missing in our two setups, I 
did it all over again in a new VM, and this time installed into 
/usr/local. It all worked, so there is some difference we're missing.

closed 2023-04-19 by louis-etne

pat-rogers, 2023-04-20:
That's great news!