AdaCore/Ada_Drivers_Library #431: Incorrect usage of 'Val instead of 'Enum_Val

liampwll, 2024-06-13:
The below line will raise a Constraint_Error if Range_Check is enabled as the enumeration is not continuous, using 'Enum_Val instead would fix this. This error could exist in the thousands of other uses of 'Val which I have not checked and it could have potentially worse consequences than this one which will likely be harmless when Range_Check is disabled.

https://github.com/AdaCore/Ada_Drivers_Library/blob/b651d01dcfd7e27836f938c1d494e1ed3212ed43/arch/ARM/STM32/drivers/stm32-adc.adb#L410

liampwll, 2024-06-13:
Here is another case of the same error, this one cal return an incorrect value:

https://github.com/AdaCore/Ada_Drivers_Library/blob/b651d01dcfd7e27836f938c1d494e1ed3212ed43/arch/ARM/STM32/drivers/stm32-gpio.adb#L439

pat-rogers, 2024-08-12:
On 13-Jun-24 2:24 AM, Liam Powell wrote:
> Here is another case of the same error, this one can return an incorrect 
> value:
> 
> https://github.com/AdaCore/Ada_Drivers_Library/blob/b651d01dcfd7e27836f938c1d494e1ed3212ed43/arch/ARM/STM32/drivers/stm32-gpio.adb#L439 <https://github.com/AdaCore/Ada_Drivers_Library/blob/b651d01dcfd7e27836f938c1d494e1ed3212ed43/arch/ARM/STM32/drivers/stm32-gpio.adb#L439>

Thank you for opening this issue.

I have not yet looked at the first example, but I believe that the code 
in this second example case is correct.

Specifically, in the following:

    return EXTI.External_Line_Number'Val (GPIO_Pin'Pos (This.Pin));

is correct because:

* type External_Line_Number does not have a representation clause, 
therefore the representation values are zero-based (ie, 0 through 22), and

* the result of 'Pos is always zero-based, so GPIO_Pin'Pos returns 0 .. 
15, and

* 0 .. 15 are valid representations for External_Line_Number values when 
converted via 'Val, and

* those numeric values result in the External_Line_Number values that 
match the corresponding GPIO pins.

Let me know if you disagree. It is always possible that I have missed 
something.

In any case I will also look at the first example in the issue.

Thanks again,

Pat Rogers

liampwll, 2024-08-12:
I have since removed that one but GitHub obviously can not update the
email, it was only problematic in a fork rather than this repo. The first
one is still an issue with Range_Check.

pat-rogers, 2024-08-12:
On 12-Aug-24 1:25 PM, Liam Powell wrote:
> I have since removed that one but GitHub obviously can not update the
> email, it was only problematic in a fork rather than this repo. The first
> one is still an issue with Range_Check.

OK, I'll take a look.

Pat

pat-rogers, 2024-08-12:
On 13-Jun-24 2:14 AM, Liam Powell wrote:
> The below line will raise a Constraint_Error if Range_Check is enabled
> as the enumeration is not continuous, using 'Enum_Val instead would fix
> this. This error could exist in the thousands of [...]

I agree that using 'Val in this line is incorrect.