AdaCore/Ada_Drivers_Library #440: Please where can get ARM Cortex M3 Drivers

sanyaade-teachings, 2025-02-07:
Hi Everyone,
I could not find the drivers for Cortex M3
We have a large number of ARM Cortex M3 we are repurposing for our student Ada coding classes
Large number of :  https://www.vexrobotics.com/276-2194.html
They are STMicroelectronics ARM® Cortex® M3

God blesses!!!

regards

pat-rogers, 2025-02-14:
You need to know which specific MCU and vendor. Just saying M3 isn't enough to narrow down what you have in that vexrobotics product.

sanyaade-teachings, 2025-03-13:
> You need to know which specific MCU and vendor. Just saying M3 isn't enough to narrow down what you have in that vexrobotics product.

I did mention that: ==> They are STMicroelectronics ARM® Cortex® M3 above (**STM32F103** board)

Anyway I have found some repositories that have support Arduino Due board and some for STM32F103
We can use those as bases for porting to our Vex Robotics Cortex M3 and board gpio mapping

Many thanks!

God blesses!!!

Regards